Broadcom/AFBR-S50-API #2: Low laser power or bad precision after porting to STM32 M4 (L4 variant)

borjaevo, 2022-06-10:
Hello, 

Im using an AFBR-S50MV85G with a Nucleo board STM32L432KC. I "succesfuly" ported the code to the board and passed the HAL tests. 
Playing with the examples, high speed or not, i have very bad measurements when object is farther than 800mm, objects farther than 1 meter shows they are at about 2 or 3, and with objects at 2 meters it could report they are about 10, 20, or 30 meters. Also the sensor becomes too hot, especially in high speed mode and eventually could stop working reporting this error:
`ERROR: DCA: Laser Failure or High Ambient Light Detected!
Continuous low reference amplitude due to laser failure or high ambient light detected!`

When objects are closer than 800mm it works ok, so it seems like a not enough power problem? 

Also, i cannot try the example at 1000hz or faster because it reports a lot of errors:
`ERROR: Integration Timeout! Elapsed 887 usec; Timeout 465 usec!`

This board only has one 32bit timer i can use for the LTC, so i'm suspecting maybe my LTC timer implementation is bad. Could a bad LTC timer implementation cause this errors? What other software issues could be causing this behavior?

I check with two different AFBR-S50MV85G and different components including the board and with both i have the same results. 
Thanks

goetzfrv, 2022-06-15:
Hi Borjaevo,
thanks for your feedback.
To get a better understanding of your issue would you pls provide a screendump of your terminal output?

Which kind of object does the sensor look at?

Additionally, you could add the following print function in your code to see the (integration) setting of the sensor
```
print("AID: %06d, DID: %03d, OP: %03d, PG: %02d, State: 0x%04X",
		  res->Frame.AnalogIntegrationDepth,
		  res->Frame.DigitalIntegrationDepth,
		  res->Frame.OutputPower,
		  res->Frame.PixelGain,
		  res->Frame.State);
```
Do you see the same issues when using the simple example (-> no PIT in use)?

You may also want to look at the single pixel results with the following implementation

```
static void print_singlePixelResults(argus_results_t const * res)
{
	print("Pixel range:\n");
	for (unsigned int y = 0; y < ARGUS_PIXELS_Y; y++)
	{
		print(" %6d %6d %6d %6d %6d %6d %6d %6d\n",
			res->Pixel[0][y].Range / (Q9_22_ONE / 1000),
			res->Pixel[1][y].Range / (Q9_22_ONE / 1000),
			res->Pixel[2][y].Range / (Q9_22_ONE / 1000),
			res->Pixel[3][y].Range / (Q9_22_ONE / 1000),
			res->Pixel[4][y].Range / (Q9_22_ONE / 1000),
			res->Pixel[5][y].Range / (Q9_22_ONE / 1000),
			res->Pixel[6][y].Range / (Q9_22_ONE / 1000),
			res->Pixel[7][y].Range / (Q9_22_ONE / 1000));
	}

	print("Pixel amplitude:\n");
	for (unsigned int y = 0; y < ARGUS_PIXELS_Y; y++)
	{
		print(" %6d %6d %6d %6d %6d %6d %6d %6d\n",
			res->Pixel[0][y].Amplitude / (UQ12_4_ONE),
			res->Pixel[1][y].Amplitude / (UQ12_4_ONE),
			res->Pixel[2][y].Amplitude / (UQ12_4_ONE),
			res->Pixel[3][y].Amplitude / (UQ12_4_ONE),
			res->Pixel[4][y].Amplitude / (UQ12_4_ONE),
			res->Pixel[5][y].Amplitude / (UQ12_4_ONE),
			res->Pixel[6][y].Amplitude / (UQ12_4_ONE),
			res->Pixel[7][y].Amplitude / (UQ12_4_ONE));
	}

	print("Pixel status:\n");
	for (unsigned int y = 0; y < ARGUS_PIXELS_Y; y++)
	{
		print(" 0x%02X 0x%02X 0x%02X 0x%02X 0x%02X 0x%02X 0x%02X 0x%02X\n",
			res->Pixel[0][y].Status,
			res->Pixel[1][y].Status,
			res->Pixel[2][y].Status,
			res->Pixel[3][y].Status,
			res->Pixel[4][y].Status,
			res->Pixel[5][y].Status,
			res->Pixel[6][y].Status,
			res->Pixel[7][y].Status);
	}
}
```

In the end please also post your timer.c implementation from the platform folder.

Thank you

goetzfrv, 2022-06-27:
Hello Borja,
can you please send a picture of your setup and target? 
Did you verify that the laser properly hits the target/object? This can be done with an IR cam (w/o IR filter).
In case there are other objects (or e.g. the sidewall) falling into the IR laser beam your result might be impacted by that.
Thanks

borjaevo, 2022-06-27:
> Hello Borja, can you please send a picture of your setup and target? Did you verify that the laser properly hits the target/object? This can be done with an IR cam (w/o IR filter). In case there are [...]

Hello goetzfrv,

If you click on this link you can watch my board setup, is pretty basic: [https://photos.app.goo.gl/RSUUt7mwsHM2b1me6](url)

But it follow the schematic shown in the guides about capacitors, ferrite bread, and everything. 

I have not sent you any picture about the object im doing the testing with, because its a white sheet of paper A4 size. Also i tried with my body and the same happens. 

Thanks

goetzfrv, 2022-06-27:
Hello Borja,
the link to the picture is not working. Please again provide a picture. Your previously wrote that you figured out that you had too long wires. You can also use the support.tof@broadcom.com account to send a picture of the setup and your layout files. 
Do you have a second sensor to double-check your HW and SW? 
thanks

borjaevo, 2022-06-28:
> Hello Borja, the link to the picture is not working. Please again provide a picture. Your previously wrote that you figured out that you had too long wires. You can also use the [support.tof@broadcom.com](mailto:support.tof@broadcom.com) account to send a picture of the setup and your layout files. Do you have a second sensor to double-check your HW and SW? thanks

Hello, I dont know if its because is an external link, but if you copy and paste de link to your address bar it works. There you can see the pictures. 
About the sensor, yes, i have two boards with two sensors and both do exactly the same.

closed 2022-06-28 by goetzfrv

ksumit12, 2024-07-25:
@borjaevo , how did you manage to port into nucleo board? i am using nucleo-g474re board but can seem to build the project.