Broadcom/AFBR-S50-API #3: Creation of Japanese README

waarrk, 2022-08-30:
I would like to use this sensor. Create Japanese version of README

goetzfrv, 2022-08-30:
Thanks Waarrk for your request. Could you briefly explain what you did and why you translated the readme again into japanese?
Is there something which we can improve in the english version? Please do let us know.
Thanks

waarrk, 2022-08-31:
Because this sensor is special, interest in it is spreading among students in Japan as well. If there is a Japanese README, I translated it because I thought it would be useful for developers who will be using this sensor in the future.

I just noticed that the English version of my commit has an automatic indentation fix... it doesn't seem to be a problem.

Regarding the English version of the README, it would be nice if the usage for Coretex on the official HP was expanded.
Best regards,

c-berger, 2022-10-04:
Closed as we can not maintain Japanese readme, see: https://github.com/Broadcom/AFBR-S50-API/pull/4#issuecomment-1235219212

closed 2022-10-04 by c-berger