Broadcom/AFBR-S50-API #5: Add Measurement Mode

AlexKlimaj, 2022-09-15:
I would like a measurement mode that takes as long as it needs to produce a valid measurement. Currently you need to set a measurement time in Argus_SetConfigurationFrameTime.

c-berger, 2022-10-04:
Hi Alex,
thanks for your suggestion.
Currently, the measurement frame time is fixed. That way, it is not easily possible to create a dynamic frame time settings as you suggest. As an alternative solution, you could add a filter that simply dismisses all measurement frames without a valid signal to achieve the same effect?

AlexKlimaj, 2022-10-04:
I set the measurement frame time. Then if the measurement doesn't return valid data, its not published. In this scenario, I'd need to slowly increase the measurement time until it starts returning data. Then increasing it until it no longer returns. 

I'd like to see the AFBR run as fast as possible where it returns the measurement as soon as its over a certain signal quality. Then it can take as long as it needs to return a measurement when conditions are difficult.

goetzfrv, 2022-10-18:
Great feedback, Currently, the SW architecture is still based on a fixed frame rate which is demanded by many industrial applications. However, there are already plans for a future SW release to implement a mode similar to the one described in your last post. Of course, everything is then going to be uploaded to GitHub as well.

closed 2023-08-13 by c-berger