Broadcom/AFBR-S50-API #7: Multiple Sensor

ArthurAubertin, 2023-01-27:
Is there a way to use two or more LIDAR sensors without having to rewrite a lot of argus file ?

c-berger, 2023-01-27:
Hi, we are currently working on exactly that feature which will be included in the next release in approximately 2-3 months from now. A sophisticated way of operating multiple devices per MCU will become available then.

Meanwhile, only a limited support for multiple devices is available. The idea is to initialize a dedicated instance of the `argus_hnd_t` for each sensor and pass an unique slave identifier to the `Argus_Init` function that is used in the SPI layer to address the corresponding device.
Once initialized, measurements can only take place in turns, i.e. a device at a time. Use the `Argus_TriggerMeasurement` function and wait until the measurement finishes before starting the next. 
Please also refer to the "simple example" on how to do so for a single device.

ArthurAubertin, 2023-03-28:
Hi,

We exchanged two months ago about the next release of your GitHub for multi-LiDAR sensor use. Would you happen to know more about this release?

Best regards

Arthur Aubertin

Ph. D. Acoustic and Communication
R&D Director
STIMSHOP
+33 1 85 64 10 63

c-berger, 2023-03-28:
Yes, we have just finished our testing and will push the update later this week.

ArthurAubertin, 2023-03-28:
Hi,

Thank you for your quick answer. We are looking forward to use it.

Best regards

Cordialement

Arthur Aubertin

c-berger, 2023-03-30:
Hi, 

I just pushed the new release:
https://github.com/Broadcom/AFBR-S50-API/releases/tag/v1.4.4

Let me know if any further information is needed.

Best regards,
Christoph Berger

closed 2023-03-30 by c-berger