Broadcom/AFBR-S50-API #8: Disabling the Pixels

ELE495-GRUP9, 2023-02-13:
We want to use the sensor with high frame rates (3kHz). According to what is written in the datasheet, we need to reduce the number of pixels to 8. We are trying to do this right now. But we got a few issues.
How can we manage the pixels safely, how safe is it to disable random pixels? . 
We found a "argus_map.h" header file in API. We thought, we can use macros in it. But, what is our 32-bit pixel mask that is used as arguments in macro functions here? How can we determine this?
And last one, can we find an example C/C++ code about disabling pixels safely?

closed 2023-02-15 by goetzfrv

cypamigon, 2024-03-28:
Hello, 

We are using an AFBR-S50MV85I and we want to artificially reduce the FoV by disabling some pixels. 
We tried to use the `PIXELXY_DISABLE(msk, x, y)` macro defined in `argus_map.h` but I'm not sure about the way to use it. 
Here is the relevant code section : 

``` c

	argus_cfg_pba_t pba;
	status = Argus_GetConfigurationPixelBinning(hnd, &pba);
	handle_error(status, "Get PBA configuration failed!");

	// Exclude pixels from the PBA to reduce the FoV
	PIXELXY_DISABLE(pba.PrefilterMask, 0, 0);
	PIXELXY_DISABLE(pba.PrefilterMask, 0, 1);
	PIXELXY_DISABLE(pba.PrefilterMask, 0, 2);
	PIXELXY_DISABLE(pba.PrefilterMask, 0, 3);
	PIXELXY_DISABLE(pba.PrefilterMask, 7, 0);
	PIXELXY_DISABLE(pba.PrefilterMask, 7, 1);
	PIXELXY_DISABLE(pba.PrefilterMask, 7, 2);
	PIXELXY_DISABLE(pba.PrefilterMask, 7, 3);

	status = Argus_SetConfigurationPixelBinning(hnd, &pba);
	handle_error(status, "Set PBA configuration failed!");

``` 

It does not appear to change anything. I've tried to disable all pixels to make it clear but it behave the same. Can you provide any guidance on how to disable pixels ? 
Thanks

c-berger, 2024-03-29:
Hi, just replace the `PIXELXY_DISABLE` with `PIXELXY_ENABLE`.
The prefilter mask determines which pixels are removed (i.e. filtered) from the bin. This is done by setting the pixels corresponding bit in the mask to `1`. Thus, you need to enable that bit in the mask which in turn removes the pixel.
Sorry for the confusion.
Have a nice weekend!