Broadcom/AFBR-S50-API #10: Improve reactivity

cypamigon, 2023-07-21:
Hi, 

I am designing a product that embed a Broadcom AFBR-S50MV85G in order to detect objects and people in the range 0 - 5 meters, in indoor and outdoor. 
I have designed the software base on your API (version 1.4.4), it work great but I would like to improve detection speed. Indeed ,when something is passing very quickly in the ToF FOV, the range measurement are not updated accordingly, only the status code returned is changing (- 110 instead of 0).

[image]

I assume this latency is due to the Dynamic Configuration Algorithm which is updating is parameters due to this suddent distance change. So here is my questions : How can we reduce this latency ? Can we disable the DCA ? Can we change the parameters used by the DCA to improve reactivity ? Thank for your answers. 

For information, I am using the sensor at 100 Hz with Short Range mode, DFM Mode 4X, and SNM in dynamic mode.

c-berger, 2023-07-21:
Hi,

thanks for your explanation. Here are a few quick tips that might help:

- if you never go beyond the unambiguous range (approx. 6 in short range mode), you could disable the DFM. The DFM requires a few consecutive frames to create a valid range. If sudden changes in distance happen, it will go out-of-sync and require a few frames to get back to valid measurement values. Without DFM, there is no need for that synchronization and each frame has a valid signal as soon as the pixel is not over- or underexposed.
- if 6m unambiguous range is not sufficient, you might want to consider the long range mode for 12m unambiguous range without DFM.
- The DFM limits the frame rate to approximately 100 fps. Without DFM you could go higher for faster response times (which is then traded for noise).
- There are a few parameters in the DCA settings that can improve the reaction time, especially from long to short integration times (which is the case if suddenly an object at 20cm appears). Especially the pixel saturation counts might be helpful.

Happily, we are currently working on exactly that issue. We are improving DCA and pixel binning algorithm to get faster and more reliable results when objects appear suddenly in front of the sensor. This will be released by end of the year.

If you have a reasonable business case, you can also reach out to our [support team](mailto:support.TOF@broadcom.com) which can help with all the details and maybe share a pre-release version with the latest improvements.

Best regards

cypamigon, 2023-07-24:
Hi, 

Thanks for all the precisions.
Unfortunately, we need the DFM mode to avoid false detections in case there is objects/people beyond the unambiguous range. 
I have tried to play around pixel saturation counts parameters by trying different values but to be honest I don't really see improvements in sensor reactivity. You will find below the method used. Is it the good way to achieve it ? Are you talking about these parameters ? 

``` ruby
	argus_cfg_dca_t dca;
	status = Argus_GetConfigurationDynamicAdaption(hnd, &dca);
	handle_error(status, "Get DCA configuration failed!");
	dca.SatPxThExp = 1; // [1;33] 
	dca.SatPxThLin = 1; // [1;33]
	dca.SatPxThRst = 1; // [1;33]
	status = Argus_SetConfigurationDynamicAdaption(hnd, &dca);
	handle_error(status, "Set DCA configuration failed!");
```

I've also tried to increase the frame rate up to 200 fps, even if I know the maximum is around 100 fps as you explain. Measurements values still seems reliables, I just have a new status code appearing some times (106). Is the frame rate internally limited by the API ? If no, can I push it above 100 fps ? What are the risks to go higher ?

c-berger, 2023-07-24:
Yes, that are exactly the settings I was talking about. Setting them to 1 would cause the DCA to reset to the minimum settings as soon as a saturated pixel occurs.

The DFM is limited by the internal PLL that needs to be de-tuned for each frame. This in turn requires the PLL to lock again before a new measurement can be carried out. This will take a few milliseconds. The frame rate of 100 fps is the fastest frame rate that leaves enough time for the measurements while ensuring that the PLL can always lock properly.
That said, there is no serious risk when increasing the frame rate above 100 fps other than loosing accuracy/reliability.
The 106 status you observe will happen any time the PLL does not lock properly.
Apart from that, the higher noise may also bring new issues with the DFM such as it randomly loosing it's sync state and raising an error (-110).

After all, it really depends on your application specific needs. You will need to test higher frame rates and see if you can afford loosing accuracy in favor of gaining reaction speed.

closed 2023-09-15 by c-berger