Broadcom/AFBR-S50-API #12: Incorrect range when entering/exiting beam

NathanSweet, 2023-08-08:
When an object enters or exits the laser, I am seeing 0-3 invalid ranges. For example (annotated afterward):

```
[many 3638 mm entries]
68 Hz, 0 status, 100 quality,   83 LSB amplitude, 3332 mm  <- incorrect
61 Hz, 0 status, 100 quality,  698 LSB amplitude, 1819 mm  <- incorrect
57 Hz, 0 status, 100 quality, 1092 LSB amplitude,  560 mm
61 Hz, 0 status, 100 quality, 1200 LSB amplitude,  582 mm
55 Hz, 0 status, 100 quality, 1129 LSB amplitude,  591 mm
56 Hz, 0 status, 100 quality,  921 LSB amplitude,  595 mm
51 Hz, 0 status, 100 quality,  699 LSB amplitude,  553 mm
64 Hz, 0 status, 100 quality, 1149 LSB amplitude,  590 mm
68 Hz, 0 status, 100 quality,  601 LSB amplitude,  593 mm
53 Hz, 0 status, 100 quality,  600 LSB amplitude,  594 mm
68 Hz, 0 status, 100 quality,  569 LSB amplitude,  588 mm
53 Hz, 0 status, 100 quality,  483 LSB amplitude,  591 mm
68 Hz, 0 status, 100 quality,  409 LSB amplitude,  590 mm
53 Hz, 0 status, 100 quality,  327 LSB amplitude,  601 mm
69 Hz, 0 status, 100 quality,  353 LSB amplitude,  579 mm
53 Hz, 0 status, 100 quality,  391 LSB amplitude,  593 mm
68 Hz, 0 status, 100 quality,  316 LSB amplitude,  978 mm  <- incorrect
52 Hz, 0 status, 100 quality,  169 LSB amplitude, 1921 mm  <- incorrect
72 Hz, 0 status, 100 quality,   98 LSB amplitude, 3375 mm  <- incorrect
[many 3638 mm entries]
```

This may be related to #10, but I didn't want to clutter up that issue if it is not.

As mentioned in #10, I tried disabling DFM. I also (separately and together) tried adjusting DCA as shown there. I did not see that this changed the problematic behavior.

```
status = Argus_SetConfigurationDFMMode(device, DFM_MODE_OFF);
checkFail(status, "Argus_SetConfigurationDFMMode failed!", true);

argus_cfg_dca_t dca;
status = Argus_GetConfigurationDynamicAdaption(device, &dca);
checkFail(status, "Get DCA configuration failed!", true);
dca.SatPxThExp = 1; // [1;33]
dca.SatPxThLin = 1; // [1;33]
dca.SatPxThRst = 1; // [1;33]
status = Argus_SetConfigurationDynamicAdaption(device, &dca);
checkFail(status, "Set DCA configuration failed!", true);
```

Is there a way to avoid these readings? If not, what approach should be used to discard them?

BTW, it'd be helpful if the doxygen API documentation listed default values.

c-berger, 2023-08-09:
Hi Nathan,

Thanks for the idea with default values, I will consider adding those...

Regarding your problem, can you elaborate on what you'd expect here? What exact sensor type are you using? Looks like it's mixing near and far distances while the object is entering or leaving the field-of-view. The pixel binning algorithm is optimized for flat surfaces across the field-of-view. This behavior can be changed but highly depends on your application specific requirements.

NathanSweet, 2023-08-09:
Sure, I'm using AFBR-S50MV85G. My use case is primarily a break beam sensor and it works perfectly for that. Now I need "gestures", so the beam can be broken by a hand, then moved closer or farther (or not at all). When the reported range values are unstable, it is easy to mistake the jump in values as a gesture. I can buffer the values and accept the latency if needed, but first is there configuration that would help?

c-berger, 2023-08-13:
Ok, thanks for the details. When you say gestures, I guess you are using mainly the region close to the sensor. There is a parallax effect (see section 3.4.3.1 in [AFBR-S50-BAS-Time-of-Flight-Basics-AN](https://docs.broadcom.com/docs/AFBR-S50-BAS-Time-of-Flight-Basics-AN)) that could cause your false positive gestures when moving the hand up and down. This is because the same reflex from an object will be projected into different pixels depending on the distance of the object to the sensor. It appears to move across the pixel field when in fact it only moves in perpendicular direction. In order to mitigate that effect, you'd need to carefully design the gesture recognition algorithm.

Anyway, as this is rather an application topic than a software issue, I'd kindly ask you to elaborate your application specific requirements to our [TOF support team](mailto:support.TOF@broadcom.com). You may add some detailed description and measurement data such that we can analyse the problem and come up with a solution. There are quite a few configuration parameters that can help to increase the stability of your measurement results too. But we'd required more information to avoid unwanted side effects.

Best regards!

closed 2023-08-13 by c-berger