Broadcom/AFBR-S50-API #13: Underwater Calibration

Lucas-DLF, 2023-08-10:
Hi,

I am currently using 3 AFBR-S50MV68B with an STM32L451RE. Those sensors perform underwater measurements.
As the light moves slower in water, I just had to add a 0.75 coefficient to the obtained the correct "Range" values.
So far it is working fine.

A few weeks ago, you released a new version including the sensor calibration. As I use cover glasses in my application I would like to perform such a calibration. However, it is impossible to perform the step 2 (Electrical + Optical Xtalk Measurement (without Cover)) as the cover glass ensures waterproofing...

I thought about two options:

1. Doing a 'classic air' calibration and hope for a correct behaviour underwater.
2. Doing a first 'classic air' calibration:
Consider the impact of the cover glass erased.
Doing a second calibration, with water measurements instead of cover glass effect. Something like:
> 1 Step1
> 2 Long distance measurement on optical sink, **in air with cover glass**
> 3 Long distance measurement on optical sink, **in water with cover glass**

Which one do you think it is the best?
Do you have any suggestions?

Thanks,
Lucas

NathanSweet, 2023-08-10:
I have limited experience with this sensor, but FWIW I am using 2mm cover glass and my sensor is pressed against the glass. Conveniently I don't need cover glass calibration because reflections off the cover glass from the transmitter (mostly?) cannot bounce into the receiver. When there is space between the sensor and cover glass then reflections can get into the receiver. I think it's better to avoid that if at all possible. Otherwise I'm of no help to your calibration questions.

Lucas-DLF, 2023-08-10:
Thanks for your answer Nathan.

Indeed, the need of xtalk calibration is less useful in this particular case.
Unfortunately in my application, I have a gap of about 0.5mm... Just above 0.1mm, where it is especially necessary.

And I sadly can't modify the air gap.

c-berger, 2023-08-13:
I guess you are referring to the recently released cover glass ([AFBR-S50-Cover-Glass-Calibration-AN](https://docs.broadcom.com/docs/AFBR-S50-Cover-Glass-Calibration-AN)) and crosstalk calibration ([AFBR-S50-XTK-Crosstalk-Guide](https://docs.broadcom.com/docs/AFBR-S50-XTK-Crosstalk-Guide)) application notes? If not, please have a look. Of course, they do not cover underwater applications. But it should provide better understanding of the topic in general.

For now, we have only limited experience with underwater applications. So you maybe just need to test the results of your suggestions. Both approaches seem reasonable to me. If you can use the optical sink underwater, it should be possible to somehow run the calibration underwater too. 

Anyway, as this is more a application support topic, please reach out to our [TOF support team](mailto:support.TOF@broadcom.com) for a better and more detailed help. I am sure we can figure out an reasonable approach.

closed 2023-08-13 by c-berger

Lucas-DLF, 2023-08-16:
Yes, I was referring to the application notes.

Thanks for your suggestions.