Broadcom/AFBR-S50-API #14: Can't run step-by-step crosstalk calibration

cypamigon, 2023-08-28:
Hi, 

I am using an AFBR-S50MV85G ToF sensor behind a cover glass and I would like to run a calibration in order to mitigate optical crosstalk from this cover glass. I am using CubeIDE for developping the software. 
I've integrated both argus_xtalk_cal_cli.h and argus_xtalk_cal_cli.c into my project and I'm invoking  ```Argus_XtalkCalibration_CLI() ``` within the  ```main() ```. However, I can't compile the project because the function  ```UART_SetRxCallback() ``` is not defined. 
I've been following [Broadcom Programming Guide](https://docs.broadcom.com/doc/AFBR-S50-SDK-Porting-Guide-to-Cortex-M4-PG) (version 1.2) to set up the UART Interface (P62) but there is no information provided regarding UART RX handling. I assume I need to establish the UART RX interface in a manner similar to the TX interface, but I'm unsure of the approach.

For now, I'm retrieving user inputs by calling ```HAL_UART_Receive_DMA(&huart2, &rxdata, 1)``` in  ```Get_UARTRxdata() ```. It seem to work, but I assume this is not the optimal solution and it might be the reason of my problem. Indeed, when I run the step-by-step crosstalk procedure  (by typing 's' in a terminal), I'm receiving weirds characters (just like a wrong baudrate ) : 

[image]

So here is my question, how can I accomplish the step-by-step calibration successfully ? 

Regards

goetzfrv, 2023-08-28:
Dear cypamigon,
thanks for your questions and raising this topic. It seems that the programming guide is not yet updated for the usage of the xtalk calibration CLI. However, there is a solution in the current repository because it also contains a working UART Rx interface for the STM project. Just clone or download and import it into your IDE. Then add the xtalk files as described in the app note and run the calibration CLI accordingly. Please let us know if your issue is resolved.

cypamigon, 2023-08-29:
Ok, I've found why the task selection in the CLI were not triggered on character reception. We need to add the call to the function `USER_UART_IRQHandler(UART_HandleTypeDef *huart)` on `USART2_IRQHandler(void)` (in **'stm32f4xxx_it.c'**) otherwise the callback is not triggered on character reception. 
So in **'stm32f4xxx_it.c'**' add : 

``` c
/* Private typedef -----------------------------------------------------------*/
/* USER CODE BEGIN TD */
extern void USER_UART_IRQHandler(UART_HandleTypeDef *huart);

/* USER CODE END TD */
```
and add the call in `USART2_IRQHandler()` : 

``` c
void USART2_IRQHandler(void)
{
  /* USER CODE BEGIN USART2_IRQn 0 */

  /* USER CODE END USART2_IRQn 0 */
  HAL_UART_IRQHandler(&huart2);
  /* USER CODE BEGIN USART2_IRQn 1 */

  USER_UART_IRQHandler(&huart2);

  /* USER CODE END USART2_IRQn 1 */
}
```

I'm still having an hardfault when i run the calibration though.

closed 2023-08-30 by cypamigon