Broadcom/AFBR-S50-API #15: E2 Lite Debugger

walkthedog2, 2023-09-12:
Has anyone been successful with connecting to the Mikroe reference board using a E2 Lite Debugger? 
(https://www.mikroe.com/bdc-afbr-s50-tof-sensor-board)

Running Projects/e2Studio/AFBR_S50_Example_RA4M2 gives this error: ` Not connected to the user system.`

````
Starting server with the following options: 
        Raw options                : C:\Users\thwai\.eclipse\com.renesas.platform_1435879475\DebugComp\\RA\e2-server-gdb -g E2LITE -t R7FA4M2AD -uConnectionTimeout= 30 -uClockSrc= 0 -uAllowClockSourceInternal= 1 -uInteface= SWD -uIfSpeed= auto -w 0 -z 33 -uResetCon= 1 -uLowPower= 1 -uEnableSciBoot= 0 -uresetOnReload= 1 -n 0 -uFlashBp= 1 -uhookWorkRamSize= 0x400 -ueraseRomOnDownload= 1 -ueraseDataRomOnDownload= 1 -uTraceMTB= 0 -uTraceSizeMTB= 1024 -uProgReWriteIRom= 0 -uProgReWriteDFlash= 0 -uOSRestriction= 0 -uTimeMeasurementEnable= 1 -uDWTEnable= 0 -uMemRegion= 0x20000000:0x4C00:RAM:s -uMemRegion= 0x20004C00:0x1400:RAM:c -uMemRegion= 0x00000000:0x20800:FLASH:s -uMemRegion= 0x00020800:0x7800:FLASH:c -uMemRegion= 0x08000000:0x0:DATA_FLASH:s -uMemRegion= 0x0100A100:0x80:OPTION_SETTING:s -uMemRegion= 0x0100A200:0xCC:OPTION_SETTING_S:s -uMemRegion= 0x00000000:0x0:ID_CODE:s -uMemRegion= 0x80010000:0x0:SDRAM:s -uMemRegion= 0x60000000:0x0:QSPI_FLASH:s -uMemRegion= 0x80020000:0x0:OSPI_DEVICE_0:s -uMemRegion= 0x80030000:0x0:OSPI_DEVICE_1:s -uMemRegion= 0x00028000:0x58000:FLASH:n -uMemRegion= 0x08000000:0x2000:DATA_FLASH:n -uTz= SSD -l -uCore= SINGLE_CORE|enabled|1|main -uSyncMode= async -uFirstGDB= main --english --gdbVersion= 12.1

Connecting to E2LITE, ARM target
        GDBServer endian           : little
        Target power from emulator : Off 
Starting target connection
GDB action 'connect to target', has failed with error report, Error 0x00040102: Not connected to the user system.

force_kill_gdb()
Disconnected from the Target Debugger.
````

Any assistance would be greatly appreciated.
Sebastian 
University of Newcastle

walkthedog2, 2023-09-13:
Resolved.
For anyone attempting the same, make sure you ground pin 9 :)
https://community.renesas.com/mcu-mpu/ra/f/forum/19094/e2-lite-connection

closed 2023-09-13 by walkthedog2

c-berger, 2023-09-13:
Hi and thanks for the info. That is surely helpful. We have not been aware of such an issue. We only have tested with a Jlink and a Renesas probe that comes with their evaluation boards...

Best regards!