Broadcom/AFBR-S50-API #16: Import issue

Dawei-Shi, 2023-09-13:
Hi,

I am trying to import this project to E2 Studio. but there were 19 errors when I built it as shown in the figure. So what can I do about it?

kind regards
Davidd
[image]

c-berger, 2023-09-14:
Hi,
what version of the e2-studio IDE and FSP package are you using?
We are currently investigation this issue as we also face it on one of our computers. We think it is some incompatibility with an older FSP version. Thus, I would recommend updating to the latest version as a fist step. Maybe you can also try to regenerate the code with your version, if an update does not help.
Please let us know if you find any solution...

Dawei-Shi, 2023-09-14:
> Hi, what version of the e2-studio IDE and FSP package are you using? We are currently investigation this issue as we also face it on one of our computers. We think it is some incompatibility with an [...]

Hi,

my e2-studio is the latest version and the FSP version is 3.6.0.
[image]
I'll try a different version, and could you give me any instructions on how to regenerate the code with my version since I am a beginner with this platform? thank you.

Regards
Davidd

c-berger, 2023-09-14:
Ok, thanks, version 3.6.0 maybe causes the issue. We have tested with version 4.4.0 of FSP. Can you download it or a newer version from here: https://github.com/renesas/fsp/releases (e.g. `FSP_Packs_v4.6.0.exe`)

After installing, you can restart the IDE and do a clean rebuild. You maybe need to remove the Release/Debug folders and restore the Git commit.

To re-generate the code, open the `configuration.xml` file in the project within e2 Studio and click on the Generate Project Content button. You can also check if it is using the correct versions here:
[image]

Dawei-Shi, 2023-09-14:
> Ok, thanks, version 3.6.0 maybe causes the issue. We have tested with version 4.4.0 of FSP. Can you download it or a newer version from here: https://github.com/renesas/fsp/releases (e.g. [...]

Hi,

Thank you for your help. I have changed the FSP version to 4.6.0. but the project still can not be built.
[image]

I am wondering if I missing something when I import this project. Thank you.

Regards
Davidd

c-berger, 2023-09-15:
Hi Davidd,

seems like there is still something wrong with the e2 Studio and FSP installation. We have found that installing only the FSP package over an existing e2 Studio installation can be troublesome.
I would recommend to completely remove all Renesas stuff including e2 Studio / FSP / Drivers / etc. Just go to Windows Apps & Features and remove everything labeled with Renesas. You may also want to remove the AFBR-S50 repository and the e2 Studio workspace under the %USERPROFILE%/e2_studio directory. Just to be sure...
Now, cleanly install the latest e2 Studio (for all users), selecting the RA Family packages. Not sure what FSP version is included, so you may need to install the latest FSP package afterwards.
Clone the AFBR-S50 repository again and open the e2 Studio and import the projects you want and hope for the best.
Sorry for the inconvenience, but the Renesas ecosystems seems to be a bit fragile. If you still have issue, you may need to contact the Renesas support.
Let me know if you have any success.
Best Regards!

c-berger, 2023-09-15:
Please note that we just pushed a new version of the Renesas e2-Studio projects which could also help with your issue. It contains improved HAL and a implementation of the Explorer App for the RA4M2 platform.

Dawei-Shi, 2023-09-18:
Hi,

Thank you for your update. I have solved this problem and the project can be built successful. And the reason of build fail is the FSP version in this project is not match the FSP version of E2-studio.
[image]
[image]
The build will success when those two FSP version matches. Thank you for your help again.

By the way, The FSP version of your updated project AFBR_S50_CANApp_RA4M2 is V4.2.0 and other two are V4.4.0. I am not sure if you miss this project, just let you know.
[image]

Kind regards
Davidd

c-berger, 2023-09-19:
Thank you for the feedback.
Yes, that was not intended and we will update the FSP for that project too.

closed 2023-09-21 by Dawei-Shi