Broadcom/AFBR-S50-API #18: 1.4.4 stack on startup is sometimes larger

AlexKlimaj, 2023-09-26:
I recently updated the PX4 AFBR driver to 1.4.4. I noticed that I started getting a random hardfault on boot and traced it to the AFBR driver sometimes using more stack causing it to overrun the amount allocated to the task.

The AFBR driver runs in hp_default. Currently I have 1872 bytes allocated to the task. After increasing that, I found that the AFBR sometimes uses 2068 on startup instead of 1876. 

Upon power cycling, it appears to be random how much stack it uses. Can you tell me why this is the case?

https://github.com/PX4/PX4-Autopilot/pull/21432

Before increasing the total stack allowed for the task. Non hardfault case.
``` 
PID COMMAND                   CPU(ms) CPU(%)  USED/STACK PRIO(BASE) STATE FD
   0 Idle Task                   34313 48.250   260/  768   0 (  0)  READY  3
   1 hpwork                        318  2.668  1324/ 2944 254 (254)  w:sem  3
   2 lpwork                          0  0.000   284/ 1576  50 ( 50)  w:sem  3
   3 nsh_main                        0  0.000  1548/ 2576 100 (100)  w:sem  3
   4 wq:manager                      0  0.000   452/ 1232 255 (255)  w:sem  3
   5 wq:lp_default                  39  0.358  1076/ 1896 205 (205)  READY  3
  62 wq:nav_and_controllers        596  5.135  1404/ 2216 242 (242)  w:sem  3
  38 wq:SPI1                      2056 17.531  1804/ 2368 253 (253)  w:sem  3
  55 wq:hp_default                  34  0.290  1872/ 1872 237 (237)  READY  3
  75 top                           292  2.526  3068/ 4056 237 (237)  RUN    3
  63 wq:rate_ctrl                  896  7.631  1404/ 3120 255 (255)  w:sem  3
  72 wq:INS0                       916  7.842  1244/ 5976 241 (241)  w:sem  3
  74 wq:uavcan                     642  5.486  2388/ 3600 236 (236)  READY  3

Processes: 13 total, 5 running, 8 sleeping
CPU usage: 49.47% tasks, 2.28% sched, 48.25% idle
Uptime: 64.611s total, 34.314s idle
nsh>
```

After increasing total stack. Does not hardfault. Low stack instance.
``` 
PID COMMAND                   CPU(ms) CPU(%)  USED/STACK PRIO(BASE) STATE FD
   0 Idle Task                     948 51.218   260/  768   0 (  0)  READY  3
   1 hpwork                          5  2.813  1324/ 2944 254 (254)  w:sem  3
   2 lpwork                          0  0.000   284/ 1576  50 ( 50)  w:sem  3
   3 nsh_main                        0  0.000  1540/ 2576 100 (100)  w:sem  3
   4 wq:manager                      0  0.000   628/ 1232 255 (255)  w:sem  3
   5 wq:lp_default                   0  0.324  1028/ 1896 205 (205)  READY  3
  75 top                             0  0.000  3068/ 4056 237 (237)  RUN    3
  38 wq:SPI1                        34 17.213  1892/ 2368 253 (253)  w:sem  3
  55 wq:hp_default                   0  0.310  1876/ 2368 237 (237)  READY  3
  62 wq:nav_and_controllers         10  5.091  1380/ 2216 242 (242)  w:sem  3
  63 wq:rate_ctrl                   15  7.600  1380/ 3120 255 (255)  w:sem  3
  72 wq:INS0                        15  7.522  1244/ 5976 241 (241)  w:sem  3
  74 wq:uavcan                      11  5.537  2388/ 3600 236 (236)  READY  3

Processes: 13 total, 5 running, 8 sleeping
CPU usage: 46.41% tasks, 2.37% sched, 51.22% idle
Uptime: 1.779s total, 0.949s idle
nsh>
```

After increasing total stack. Does not hardfault. High stack instance.
```nsh> top
 PID COMMAND                   CPU(ms) CPU(%)  USED/STACK PRIO(BASE) STATE FD
   0 Idle Task                    2346 48.122   260/  768   0 (  0)  READY  3
   1 hpwork                         91  2.725  1324/ 2944 254 (254)  w:sem  3
   2 lpwork                          0  0.000   284/ 1576  50 ( 50)  w:sem  3
   3 nsh_main                        0  0.000  1548/ 2576 100 (100)  w:sem  3
   4 wq:manager                      0  0.000   580/ 1232 255 (255)  w:sem  3
   5 wq:lp_default                  10  0.306  1076/ 1896 205 (205)  READY  3
  75 top                            79  2.537  3068/ 4056 237 (237)  RUN    3
  38 wq:SPI1                       586 17.516  1908/ 2368 253 (253)  w:sem  3
  55 wq:hp_default                   9  0.298  2068/ 2368 237 (237)  READY  3
  62 wq:nav_and_controllers        174  5.214  1404/ 2216 242 (242)  w:sem  3
  63 wq:rate_ctrl                  257  7.684  1412/ 3120 255 (255)  w:sem  3
  72 wq:INS0                       260  7.783  1244/ 5976 241 (241)  w:sem  3
  74 wq:uavcan                     187  5.576  2380/ 3600 236 (236)  READY  3

Processes: 13 total, 5 running, 8 sleeping
CPU usage: 49.64% tasks, 2.24% sched, 48.12% idle
Uptime: 4.714s total, 2.347s idle
nsh>```

c-berger, 2023-09-27:
Hi Alex,

I am not aware of such a issue. The startup sequence should be deterministic. But since there is real hardware involved, it could happen that the path through the code is slightly different depending on what values are obtained from the device. For example, the PLL lock time may vary slightly each time resulting in more or less repetitions when polling its state via SPI.

Just to make sure, when you talk about startup, you mean the `Argus_Init` function? Further, when you say random, everything stays the same, i.e. same configuration, same device, etc.?
Does this happen for a specific device only (which one?) or can you reproduce it with multiple devices (also, which ones?)? 
How did you measure the stack? Simply by checking what memory has been changed after calling `Argus_Init`? Just wondering if it could have the same stack usage but sometimes not using the allocated space...

Best regards,
Christoph

AlexKlimaj, 2023-09-27:
On the same device through different power cycles, the stack will vary on startup. The behavior is the same across multiple units.

I only noticed this after upgrading to 1.4.4. It might be occurring on 1.3.5 but didn't cause a hardfault because it used less overall stack.

Stack is measured in nuttx with top.

c-berger, 2023-09-28:
Ok, thanks for the information.
Is it possible to simply increase the stack size to get rid of the issue?

goetzfrv, 2025-03-28:
Hi Alex, I consider this item as being resolved with newer releases. Therefore I will close this issue. 
Please re-open if you think there are still issues in this regard.