Broadcom/AFBR-S50-API #19: AFBR-S50 Explorer

sehowald, 2023-10-03:
Hello,

I've download and install AFBR-S50 Explorer to use it wit the dev kit from broadcom but the software start and get and imediate error.

Even before connecting the board to the computer. Can you help me with that ?

Thanks
Sébastien

[image]

goetzfrv, 2023-10-04:
Hello Sébastien,
thanks for your feedback. Here are a few questions for further clarification:
Can you specify your operating system? 
Did you change the destination folder of the installation path?
Did you already try to start the installer to repair the installation?
Were there any other abnormalities during the installation?
Regards,
Volker

c-berger, 2023-10-04:
Hi Sébastien,
sorry for that inconvenience. I have the suspicion that is is due to region settings in Windows. Unfortunately we have an issue that happens when the region settings are not English. Can you test changing that settings to English (United States) (in Windows Menu, type "region" and open "Region Settings", then change the "Regional Format" to "English (United States))

If that helps, I would highly appreciate if you tell me your original region setting such that I can resolve that issue for our next release.

sehowald, 2023-10-04:
Hello,
Thanks you both for your informations and comments.
I’m on windows 11 professional. I didn’t change the folder and try to repair even re install the software.

I change the window region setting to United States and it works. I was set in Switzerland (French).

Thanks for your help 
Sebastien

c-berger, 2023-10-05:
Thanks for the feedback and I'm glad it works. We will certainly fix that issue in the next release...

Best regards!

closed 2023-10-05 by c-berger