Broadcom/AFBR-S50-API #20: AFBR-S50MV85I and Arduino

FloVesuv, 2023-11-30:
Hello,

I'm new with this kind of TOF sensor. 
Can it be controlled by an Arduino ?
Is it complicated to port the code ?

Thank you.

NathanSweet, 2023-11-30:
You can use Arduino libraries (I have not tried it, I use the STM32 HAL because that's what the AFBR-S50 examples use) but not Arduino hardware. The driver is provided as precompiled libraries only for ARM Cortex M0+ processors. It cannot be ported to work on other hardware because the driver source is not provided (and will never be, as it's the valuable magic sauce).

The easiest way to get it running would probably be to use a Nucleo or Blackpill dev board.

goetzfrv, 2023-11-30:
Hi both, thanks for your questions & comments on that. Our API is actually compatible to all ARM-Cortex-M architectures even though we do not have a 'ready-to-go' porting for an Arduino or Microchip MCU, respectively. So far we have portings for NXP, STM and Renesas MCUs. 
You can find a general porting guide under https://broadcom.github.io/AFBR-S50-API/porting_guide.html
or an exemplary step-by-step guide under https://docs.broadcom.com/docs/AFBR-S50-SDK-Porting-Guide-to-Cortex-M4-PG

FloVesuv, 2023-12-15:
Hi, thanks for your feedbacks.

I've never work with STM32 MCU, but it seems to be the better option if the AFBR-S50 examples use it.

Do you have one STM32 MCU to recommend me ? The smaller but powerful enough to support the API ? Maybe a development kit ?

Thanks for your answer.

c-berger, 2023-12-18:
Hi FloVesuv,

which MCU you should use really depends on your individual application requirements. However, if you want to make your life easy, you can simply use one of the MCUs for which we are already providing working code for. 
If you prefer a STM32 MCU, use the STM32F401RE or its corresponding evaluation board ([Nucleo-F401RE](https://www.st.com/en/evaluation-tools/nucleo-f401re.html)). We have base our [brand new family evaluation kit](https://www.broadcom.com/products/optical-sensors/time-of-flight-3d-sensors/afbr-s50-fek) on it. But any other STM32 Cortex-M based MCU will do and porting within the STM32 MCU family should be straight forward. Alternatively we also have code for NXP MCUs (MKL46z on our [single device evaluation kits](https://www.broadcom.com/products/optical-sensors/time-of-flight-3d-sensors/afbr-s50lv85d-ek)) and Renesas MCUs (RA4M2 on the [MikroE sensor board](https://www.mikroe.com/bdc-afbr-s50-tof-sensor-board)) available. Using those or similar should also be rather easy in terms of porting. But any other Cortex-M based MCU will work, however, it is more effort when it comes to the initial porting. 

You can find an overview of available MCUs here: https://broadcom.github.io/AFBR-S50-API/getting_started.html#gs_build

Also check the porting guides to understand what is required to use another platform: 
https://broadcom.github.io/AFBR-S50-API/porting_guide.html
https://www.broadcom.com/products/optical-sensors/time-of-flight-3d-sensors (select any Sensor and go to "Documentation" -> "Programming Guides")

FloVesuv, 2023-12-19:
Hi c-berger,

As I said, I don't know which MCU I'll use, I don't have a lot of individual requirements.

I've ordered the Evaluation kit unit with MKL46z MCU (AFBR-S50LV85D-EK) to start testing the ToF sensor.
If I have good results with this sensor, I need to program the MCU with my user application. 
Is it possible to reprogram the AFBR-S50LV85D-EK with a custom software, or USB ports are just used for communication ?

If not, I'll use a MCU evaluation board to develop my user application.

- If I want to use the STM32F401RE and its corresponding evaluation board [Nucleo-F401RE), where can I find all files to program it ? It seems they're not in the directory Program Files (x86)\Broadcom\AFBR-S50 SDK\Device\Projects

- If I want to use MKL46z or RA4M2, I need to do a porting, that's right ? But for the MKL46z MCU, all the files seems to be in the directory Program Files (x86)\Broadcom\AFBR-S50 SDK\Device\Projects ?

Thanks for your great support.

Regards,

c-berger, 2023-12-19:
Sure you can build your own application with all our evaluation kits. All you need is provided in this repository. No additional porting required.

Though the projects are also delivered via the SDK installer (which usually installs into the program files folder), I recommend using the code provided in this repository. 

Please read the getting started guide on how to compile and flash the example projects. The examples are extensively commented with explanations and should be a good starting point for our own ideas.

https://broadcom.github.io/AFBR-S50-API/getting_started.html

FloVesuv, 2023-12-20:
Hi,

Thank you. 

I'm sorry but I'm also new on Github, so I don't know exactly how to proceed.

I'll use my evaluation board with the MKL46z MCU. If I want to get the project from Github, I need to download it on this page ?

[image]

But the .zip file is empty.
Normally there's one project for each MCU ?

I'm sorry but everything (Broadcom sensor, Github) is new for me.

Thanks for your support.

c-berger, 2023-12-20:
Well, either you clone it using Git or, if you are not familiar with Git, you can use the "Download ZIP" button to download everything as a ZIP file...

FloVesuv, 2023-12-20:
That's what I told you in my previous message, when I download the .zip file, I've got an error, folder is empty : 

[image]

Thank you

c-berger, 2023-12-20:
Sorry for the inconvenience. But I can not reproduce that issue. I just downloaded the zip file and obtained a valid file. Not sure what went wrong here. Is the zip file of any reasonable size? Mine was about 10M.

FloVesuv, 2023-12-20:
404 KB from my side

[image]

FloVesuv, 2023-12-20:
Do you have the same or not ?

c-berger, 2023-12-20:
Well, I really don't know what's going on there. Sorry!
Please try to clone the repository by using git or maybe you could also try to get some help directly from GitHub.

PrashastiNama, 2024-01-01:
Hey! I just started working on AFBR-S50 using F401RE board. I have successfully used and implemented the porting guide. Now I want to calibrate my sensor for crosstalk but I'm unable to do so. I am using this function
`Argus_XtalkCalibration_CLI(argus_hnd_t * hnd)`
before calling this function the handle to the sensor works fine but this is not taking proper value for the handle. What am I supposed to do?

c-berger, 2024-01-08:
Thanks for the update, that's nice to know...

closed 2024-01-08 by c-berger

FloVesuv, 2024-02-13:
Hi c-berger and NathanSweet,

I've downloaded files on Github, but I have issues : 

- On MCUXpressoIDE, when I'm trying to import "AFBR_S50_Example_KL46z" from "AFBR-S50-API-main" folder ;
            - If I don't copy it in my workspace, I've this issue when building the project : 

[image]

            - If I copy it in my workspace, I've this issue when building the project : 
           
[image]

- I'm trying to use files from C:\Program Files (x86)\Broadcom\AFBR-S50 SDK\Device\Projects.  I'm trying to execute simple_example from AFBR_S50_Example_KL46z, so I have to modify selector in examples.h : 

`/*! Selector for example:
 *  - 1: 01_simple_example.c: Runs measurements in simplest blocking manner.
 *  - 2: 02_advanced_example.c: Starts measurements automatically from timer interrupt.
 *  - 3: 03_high_speed_example.c: Runs measurements with up to 3000 frames per second.
 *  - 4: 04_multi_device_example.c: Demonstrates the usage of multiple devices per MCU. */
#ifndef API_EXAMPLE
#define API_EXAMPLE **1**
#endif`

But I have this issue on my serial port viewer : 

[image]

Any idea ?

Thanks for your support.

c-berger, 2024-02-13:
Hi FloVesuv,

please do not select the "copy into workspace" option when importing as it will not copy all files and the references will get broken.
If you import without that option, all should work as expected. I have seen the error you mention but I don't now the solution. Maybe you can re-import the project and maybe right-click on the project and select "refresh". Also a "clean build" could help. Maybe you even delete the Release/Debug folders in the project to be sure everything is clean.

Regarding the example, I wonder what version of the evaluation board you have. Can you try to change the `SPI_SLAVE` to `5` in the `example.h` header file? Very old legacy boards may require to use slave 5 instead of 1.
If that does not help, you may need to double check the connection. Does the board work with the AFBR-S50 Explorer?

FloVesuv, 2024-02-13:
Hi c-berger,

Thanks for your quick answer.

I've tried again without selecting "copy into workspace", and I follow your recommendations (Refresh, clean build), but same issue : 

[image]

Is there a specific place on the laptop to store the file "AFBR-S50-API-main" from Github ? 

If I import directly the .zip file, I have no access to "copy into workspace", the box is checked by default.

Regarding the example (I'm using example from  C:\Program Files (x86)\Broadcom\AFBR-S50 SDK\Device\Projects), after your modification, it works.

Thank you.

c-berger, 2024-02-13:
I remember that the issue happens when the project is not properly imported. There are many ways to import a project and some seem to not work properly. 
Did you use the "Import existing projects into workspace" option? That's how I usually do it.
Here is a link for an article that seems to cover your issue. Maybe this is of any help?

https://community.nxp.com/t5/Kinetis-Design-Studio/No-Rule-to-Make-Target-error/m-p/320170#M303

FloVesuv, 2024-02-14:
Hi,

The issue seems to come from here :

[image]

[image]

I've tried with files located in C:\Program Files (x86)\Broadcom\AFBR-S50 SDK\Device\Projects\AFBR_S50_Example_KL46z, but it doesn't work.
I've tried to copy these folders (App, Platform and Utility) in the same folder than my project, and then change paths of Linked Ressources in the project properties. It doesn't work.

Have these files been uploaded to Github at the same time as the project?

Thank you

c-berger, 2024-02-14:
Ok, just to make sure: have you unzipped the downloaded zip file from GitHub before importing the project?

The path of the "App", "Platform", "Utility" are broken, as you can tell from the exclamation mark on those folders. 

Right click on those folders and select Properties. Here you can check the actual Location under Resource. Double check the Resolved Location and maybe edit to the correct location if it is not valid.
[image]

Also go to the C/C++ Build" and uncheck "Exclude resource from build" for all configurations (i.e. Debug and Release).
[image]

Further double check if those folders are not excluded form the build by right click on the folder and select "Resource Configuration" and "Exclude From Build", then deselect all. 
[image]

After all, the exclamation mark must vanish an all folder, otherwise the files are not found.
Hope this helps! Sorry for the inconvenience, I am not sure what has happened here. Usually the importing does work very well.
Best regards.

FloVesuv, 2024-02-14:
c-berger,

Yes, I have unzipped the downloaded zip file from GitHub before importing the project.

I've solved the issue ; I've copy the file on the Desktop, and then it works.
In other locations, it doesn't, and I don't know why. However, I followed your instructions.

Regards,

c-berger, 2024-02-14:
Ok, thanks for the update. 
Maybe there are insufficient access rights on the unzip location ...

FloVesuv, 2024-03-05:
Hello c-berger,

I'm trying to modify the AFBR_S50_Example_KL46z, 02_advanced_example.c file.
In this example, we display this distance :

`print("%5d\n", res->Bin.Range / (Q9_22_ONE / 1000));`

1st question, what measure does this instruction correspond to? `Q9_22_ONE / 1000`
All pixels ? An average ?

2nd question, if I want to get the value of each pixel independently, what are the instructions ?

In which section on your website (https://broadcom.github.io/AFBR-S50-API/example_app.html) can I find help for these questions ?

Thanks for your feedback.

Regards,

c-berger, 2024-03-06:
The `res->Bin.Range` is the 1D range values as determined from all available pixels via an algorithm we call "Pixel Binning Algorithm". It is in fixed-point format, namely `Q9.22` (see https://en.wikipedia.org/wiki/Q_(number_format)), which determines it has 22 fractional bits. In order to obtain the floating-point equivalent, one need to divide the value by `2^22` or `1<<22`. Because the formatted print expects and integer value, it is additionally converted from meter to millimeter by dividing with `1000`.

If you want to obtain some information about the individual values of the `argus_results_t` structure, use the "Go To Definition" feature of your IDE and read the corresponding doc-strings. Or you use the API reference manual: 
https://broadcom.github.io/AFBR-S50-API/structargus__results__t.html

That said, it should be quite easy to change the code to put individual pixel range values. Simply replace `res->Bin.Range` by `res->Pixel[x][y].Range` with `x` and `y` are the corresponding indices of the pixel. Please note that the quality of the individual pixels highly depend on the used device type as well as the current measurement scenario. So you better double check the according `res->Pixel[x][y].Status` flags.

FloVesuv, 2024-03-06:
Thank you for your answer, that's exactly what I need.

I have one other question : is this sensor suitable for short-distance measurements (ten centimeters or less)?
According to the datasheet, the minimum measurement range is 10mm, but when I do tests, I get negative values when my measurement is around 50mm (for a 1D measurement).
Should I collect the values of the 32 pixels independently, and only use those with a consistent value? And also get the state of the status flag ?

c-berger, 2024-03-07:
The last release v1.5.6 should have a lot of improvements regarding near-range applications. Migrating from 1.4.4 should be straightforward, but also check the [migration guide for details](https://broadcom.github.io/AFBR-S50-API/migration_guide.html#migration_guide_1_5). That should mitigate most of the issues you're facing.
If you have more questions regarding your application, please reach out to support.tof@broadcom.com for more appropriate help on those topics!

FloVesuv, 2024-03-12:
Hi c-berger,

I'm following the procedure for the calibration described page 8 of AFBR-S50-XTK-AN100.pdf  (4 steps using the CoverglassXtalkcalibration_template file ).  
I'm using the 1.5.6 version for the API, but I've download the v1.4.4 excel template (not available for the latest version).
I don't know if it's matter.

I joined the excel template fill with my 3 calibrations values : 
[
[CoverglassXtalkcalibration_v1.4.x_template.xlsx](https://github.com/Broadcom/AFBR-S50-API/files/14570852/CoverglassXtalkcalibration_v1.4.x_template.xlsx)
](url)

But for the last step, results I get look "empty", and are not recognized by the software when I want to "Import Vector Table from File".

[image]

[image]

Any idea ? Something's wrong in my procedure ?

Thank you

FloVesuv, 2024-04-08:
Hello,

I've an issue when I'm trying to connect the evaluation board with the SDA connector ; I've got this message : 

[image]

I don't know why, I didn't have issue in the last days.

Thank you

c-berger, 2024-04-09:
Hi, sorry for that inconvenience. I have also experienced instabilities using the OpenSDA/PEMicro debug utilities. Unfortunately I have not solution other than reconnecting/restarting the IDE/restarting the OS. Somehow this always solve the problem for me. If you're still not able to connect, you could double check if everything is installed properly via the device manager. Also check if nothing has changed (which sometimes happen if you start using a different USB port). 

Finally, I recommend using an external probe like a JLink as this is much more stable and versatile. If you do not have one at hand, the [JLink firmware](https://www.segger.com/products/debug-probes/j-link/models/other-j-links/opensda-sda-v2/) could also be an option, but make sure you check their terms-of-use...

FloVesuv, 2024-04-10:
Hi,

I tried many times to restart my laptop, reconnecting, restarting the IDE, I have always the same issue.
I tried also with different USB ports, same thing.

Could it be a hardware issue ?

I've read different topics with the same issue. 
I tried to download the latest OpenSDA firmware, but I couldn't succeed.
Could you send it to me ? What is the procedure to install it ?

Thank you

FloVesuv, 2024-04-15:
c-berger,

Sorry to ask again, but any idea to help me ?

I've also post a message on community.nxp, no answer to help me.

Thanks for your feedback.

Regards,

c-berger, 2024-04-15:
Hi and sorry for the delay!
I am not sure if that is a hardware issue, maybe it is. Does it show any reaction when you connect it? Does it show as FRDM-KL46z drive in the file explorer? Can you still manually copy binaries onto this drive?

This would be the link for the JLink alternative firmware: https://www.segger.com/downloads/jlink/#MCU-Link

FloVesuv, 2024-04-15:
Hi,

Yes, I've the "classical" song when I connect it and it appears in the file explorer and it the device manager : 

[image]
[image]
[image]

What files must I copy onto this drive ?

Thank you

c-berger, 2024-04-15:
What does the "laststat.txt" file contain?

Best regards,
Christoph Berger

-- 
This electronic communication and the information and any files transmitted 
with it, or attached to it, are confidential and are intended solely for 
the use of the individual or entity to whom it is addressed and may contain 
information that is confidential, legally privileged, protected by privacy 
laws, or otherwise restricted from disclosure to anyone else. If you are 
not the intended recipient or the person responsible for delivering the 
e-mail to the intended recipient, you are hereby notified that any use, 
copying, distributing, dissemination, forwarding, printing, or copying of 
this e-mail is strictly prohibited. If you received this e-mail in error, 
please return the e-mail to the sender, delete it from your computer, and 
destroy any printed copy of it.

FloVesuv, 2024-04-15:
It contains « Ready. »

FloVesuv, 2024-04-16:
c-berger,

I only have these files on the Debug folder, and I have no Release folder : 

[image]

Do I just copy the *.bin and *.axf files?

c-berger, 2024-04-16:
Hi, yes, exactly those files. Both should do as they are the same binary files. Try the *.bin first. The *.axf just contains additional debug information.

If the LASTSTAT contains Ready, it looks good to me. So it is probably not an hardware issue. 
Maybe you need to reinstall the [PEMicro drivers](https://www.pemicro.com/opensda/) (only if you have not switched to the JLink version)?
Or even you try to reinstall the MCUXpresso? 
I am sorry, but I have no clue why you still get that error with the debugger. 
Maybe you can also delete all debugger configuration files (i.e. the *.launch files in the project) and create the debug configuration from scratch? 
Have you tried to use a different board?
Have you checked the output in the console window? Probably reading through that carefully can give you additional hints. Note that you may need to select the corresponding debugger output like so:
[image]

FloVesuv, 2024-04-16:
I copy the *.bin first, but nothing happens. What should happen ?

I reinstall PEMicro drivers, same issue.
I didn't try to reinstall MCUXpresso yet.
I've deleted all debug configuration files in the project, it creates a new one (1). Same issue

I don't have another board.

[image]

[image]

[image]

c-berger, 2024-04-16:
Well, as already mentioned, it should flash that binary to the MCU. Does it?

What's the output of the debugger console? The output of the build console doesn't help much...

FloVesuv, 2024-04-16:
Yes, it seems flash the binary to the MCU, LED on the board blink during time I'm copying *.bin on the drive.

Nothing appears in the Debugger Console ....

[image]

c-berger, 2024-04-16:
Sorry for being unspecific. Actually I meant the "Console" window where you can select the "GDB PEMicro Interface Debugging". It should contain the actual error and give you hints what's going on:

[image]

FloVesuv, 2024-04-16:
I cannot select the "GDB PEMicro Interface Debugging"

[image]

c-berger, 2024-04-16:
Aha, that's probably a hint that the wrong debug interface is used. Did you select the PEMicro probes when creating a new debug launch configuration?
[image]

FloVesuv, 2024-04-16:
Yes, I select PEMicro probes when I create a new debug launch configuration :

[image]

[image]

c-berger, 2024-04-16:
Ok, whats the output of the RedLinkServer? Not sure if this is the problem here, but for me it does not use the RedLink...

FloVesuv, 2024-04-16:
I'm sorry, but I don't know what you mean (I'm new on MCUXpresso).
 How can I check the output of the RedLinkServer ?

c-berger, 2024-04-16:
Just select it here:

> I cannot select the "GDB PEMicro Interface Debugging" [...]

FloVesuv, 2024-04-17:
c-berger,

- When I tried to compie the project this morning, this error displayed : 
-
[image]

I didn't modify my software since yesterday....

- I cannot select RedLinkServer anymore ...

[image]

It's getting frustrating, dealing with all these complex issues...

FloVesuv, 2024-04-17:
I will follow your advice and uninstall and reinstall everything, I will keep you updated. 
It was not a reproach, quite the contrary, I still wanted to thank you for all the help you have given me since the beginning.

FloVesuv, 2024-04-25:
Hello,

First of all, good news: my connection problem with the evaluation board has been solved (I've contacted NXP MCUXpresso support).
So I was able to restart my tests.

I'm using a cover glass in my application, so I need to proceed to the Xtalk calibration.

I've read that there's 2 ways to save Xtalk Data, one by saving Xtalk Data to Nonvolatile Memory and one by hard-Coding the
Xtalk Vector Table.

The first is the one that interests me, but I have several questions:

- Can I save the values on the FLASH memory in the MCU? How can I use them afterwards?
- How can I reduce the size of the data, given that I'll only be taking measurements at close range (ten or so cm)? Is the DFM useful in my case?
- How can I ensure that this calibration step doesn't have to be repeated every time the MCU is restarted?

Another question: how can I integrate the Range Offset Calibration function into my code? 
And same question than previously, how can I ensure that this calibration step doesn't have to be repeated every time the MCU is restarted?

[image]

Thank you for your feedback.

c-berger, 2024-04-25:
Hi there,

great to hear you resolved that issue. Would you be so kind and let us know how you did that? It could help us to improve and it would be highly appreciated!

Regarding your questions:

To enable the flash aka. non-volatile memory storage, the only thing you need to do is implement the NVM HAL on you platform.
Have a look in the [API reference manual (NVM section)](https://broadcom.github.io/AFBR-S50-API/group__argus__nvm.html) for information on that module. You only need to implement the [HAL_ReadBlock](https://broadcom.github.io/AFBR-S50-API/group__argus__nvm.html#ga8a2466c2cf2e2651b45298703f936888) and [HAL_WriteBlock](https://broadcom.github.io/AFBR-S50-API/group__argus__nvm.html#ga6a377265320f4a026896bf3718da3e99) functions. The API will pass it's data once it likes to store it to the `HAL_WriteBlock`. Upon initialization, it will attempt to read back the information via the `HAL_ReadBlock` function. So make sure the data passed is persistent across MCU restart.

I hope this answers your questions. 

Btw. your thread has become a bit messy and we would highly appreciate if you could ask upcoming questions in a dedicated issue.
Also note that I can only answer questions regarding the code here. If you have application specific topics, you better should reach out to [support.tof@broadcom.com](mailto:support.tof@broadcom.com). Thanks!

FloVesuv, 2024-04-25:
I had to update the firmware of OpenSDA to J-link (no external J-link debugger needed)
Important thing, find a Win7 system PC to update.

Refer to : https://www.nxp.com/design/design-center/software/sensor-toolbox/opensda-serial-and-debug-adapter:OPENSDA#FRDM-KL46Z

[image]

Ok, I'll open a new topic. But for me, all the previous questions were about the code.

Regards,