Broadcom/AFBR-S50-API #24: Unable to Build Example App

hurstk2, 2024-04-23:
Hi, this is a follow up to issue https://github.com/Broadcom/AFBR-S50-API/issues/23 posted last Thursday. FSP and e2studio version are still the same.

I have downloaded the updated repository and the project now fails to build even when RUN_XTALK_CALIBRATION is set to 0.

The errors from three attempts to run when RUN_XTALK_CALIBRATION is set to 0 can be found below:
[image]
[image]
[image]

The errors from three attempts to run when RUN_XTALK_CALIBRATION is set to 1 can be found below:
[image]
[image]
[image]

The project was cleaned and refreshed between each attempt.
Building the Can App also fails with similar errors when it succeeded before the update.

hurstk2, 2024-04-24:
I have recloned the version from this commit https://github.com/Broadcom/AFBR-S50-API/commit/421a55e2f9022c3b6d209047d5a18926a16773c8 which is the version I previously had and both the CAN App and the Example App work as they did before

CAN APP Build Output:
[image]
[image]

Example App Build Output:
RUN_XTALK_CALIBRATION 0
[image]
[image]
RUN_XTALK_CALIBRATION 1
[image]
[image]

______________________________________________________________________________________________________________________________________
I decided to run some tests to see which version the CAN APP and Example App stopped building for me.

The first commit when the Example app fails to build is https://github.com/Broadcom/AFBR-S50-API/commit/390912d103caa6966e45f70f700b04d91f1453e1

https://github.com/Broadcom/AFBR-S50-API/commit/390912d103caa6966e45f70f700b04d91f1453e1 Example App build output:
[image]

I found that if I went to https://github.com/Broadcom/AFBR-S50-API/commit/2122864d245efed96f3840d89fd68b5aa49545dc browse files and cloned the repository from there the CAN App successfully builds so I decided to try recloning the main branch and the CAN App successfully built with no errors but the version I got yesterday still fails to build.

The three attempts to build the Example App from the new clone of the main branch still failed with the following errors:
[image]
[image]
[image]

c-berger, 2024-04-25:
Hi there, thanks for tracking down the commit that introduces the issue. Seems like the `Q11_4_ONE` is not defined in those projects. That should be easy to solve. I will take care of that immediately and let you know once it's fixed.

c-berger, 2024-04-26:
Hi, seems like a type has caused the issue. Oddly that I didn't recognize that in the first place.
May you please try the latest main commit? It should fix this problem...

hurstk2, 2024-04-26:
Thank you so much for your help, both the CAN App and the Example App now build successfully.

CAN APP Build Output:
[image]

Example App Build Output:
RUN_XTALK_CALIBRATION 0
[image]
RUN_XTALK_CALIBRATION 1
[image]

closed 2024-04-26 by hurstk2