Broadcom/AFBR-S50-API #25: Timer interval in 02_advanced_example.c

FloVesuv, 2024-05-06:
Hello,

By default, the timer interval for the interrupt is set to 1ms, isn't it ?

[image]

If I want to change it, I just have to modify the value here ?

Thank you.

Regards,

c-berger, 2024-05-07:
The mentioned function originates from the HAL Self Test.
I am not sure what you want to achieve with that change?
If you intend to update the measurement frame rate, you need to use [Argus_SetConfigurationFrameTime()](https://broadcom.github.io/AFBR-S50-API/group__argus__cfg.html#gaa6d970c42ac21e56907247f58a029472) to update via the API. For example, see the the `InitializeDevice` in the [advanced example](https://broadcom.github.io/AFBR-S50-API/04_multi_device_example_8c-example.html) which changes the frame rate to 10 Hz.

FloVesuv, 2024-05-07:
I don't want to modify this function, I was just looking for where to modify this value. So I start to look at functions with Timer_SetInterval().

But this is exactly the answer I needed, thank you.

Regards,