Broadcom/AFBR-S50-API #29: Is a jumper or setting needed to enable UART comms with the Explorer app?

mfreed7, 2024-05-22:
So I am using [the MIKROE-5631 reference board](https://www.digikey.com/en/products/detail/mikroelektronika/MIKROE-5631/21851068), and I followed [these instructions](https://broadcom.github.io/AFBR-S50-API/reference_board.html#reference_board_bootloader) for activating the bootloader and flashing a program. I flashed [AFBR-S50-RA4M2-v1.5.6.zip from here](https://github.com/Broadcom/AFBR-S50-API/releases/download/v1.5.6/AFBR-S50-RA4M2-v1.5.6.zip). That all works **great** when connected to the [AFBR.S50.SDK.v1.5.6-basic.msi explorer app](https://github.com/Broadcom/AFBR-S50-API/releases/download/v1.5.6/AFBR.S50.SDK.v1.5.6-basic.msi) **via a USB cable**. The whole thing talks and works very well. However, when connected **via UART**, either directly to my application circuit, or to the computer via [this FTDI USB to serial adapter](https://www.amazon.com/gp/product/B07BBPX8B8/ref=ppx_yo_dt_b_asin_title_o00_s00?ie=UTF8&psc=1), I never get any responses at all from the reference board. Not even a NAK (0x0B) response. The explorer app shows an immediate "I can't connect" error message as soon as you try to connect.

I've verified (via oscilloscope) that the transmit pulses are clean and correct, and I've tried both 3.3V and 5V. I've used the pinout provided [right here](https://broadcom.github.io/AFBR-S50-API/reference_board.html#:~:text=Serial%20(UART)-,Connector,-The%20board%20features), which seems to be correct based on pin 3 (TXD) being pulled high by the reference board. I've used my own application software, the [provided python code](https://broadcom.github.io/AFBR-S50-API/explorer_app.html#:~:text=Python%20Example%20on%20Using%20the%20SCI%20Interface), and the AFBR Explorer app. In none of these cases does any UART packet ever get transmitted back from the reference board.

I'm posting here because I'm kind of out of things to try. There must be something I'm missing about how to switch out of USB and into UART mode or something. Please help!

c-berger, 2024-05-22:
Hi and thanks for the extensive information. Unfortunately, I just realized that we simply did not provide the UART enabled firmware yet. The ExplorerApp firmware variants in the linked *.zip is for USB only. 

Sorry for the inconvenience! I will update with a UART version as soon as I am back in the office beginning of June.

Meanwhile, you can try to compile your own version of the firmware using the Renesas e2-Studio IDE as described here:
https://broadcom.github.io/AFBR-S50-API/e2studio.html

Note that you need to load the [AFBR_S50_ExploerApp](https://broadcom.github.io/AFBR-S50-API/explorer_app.html) instead to compile the firmware required for the Explorer GUI.
Once loaded, you can open the project settings and change the `AFBR_SCI_USB=1` to `AFBR_SCI_USB=0` (in the compiler settings) to disable the USB and enable the UART instead.

In case you want to spin your own communication protocol via UART anyway, you may also want to consider using the [AFBR_S50_Example](https://broadcom.github.io/AFBR-S50-API/example_app.html) app wich simply streams measurement data as formatted strings via the UART and go ahead from there.

mfreed7, 2024-05-24:
Oh great! Thanks for the quick response. That explains what I’ve been seeing. If you’re planning to build a UART-enabled version in the beginning of June, perhaps I’ll just wait for that. Would you mind pinging this issue once there’s a version to try?

c-berger, 2024-05-24:
Sure, I'll let you know...

c-berger, 2024-06-03:
Hi, I just added the missing binary to the [RA4M2 v1.5.6](https://github.com/Broadcom/AFBR-S50-API/releases/download/v1.5.6/AFBR-S50-RA4M2-v1.5.6.zip) archive. It now contains two version of the ExplorerApp, one for USB and one for UART. Please let me know if this fixes your issue.

Side note: The [AFBR_S50_ExploerApp](https://broadcom.github.io/AFBR-S50-API/explorer_app.html) projects has a `Release` and `ReleaseUART` configuration prepared to build the `USB` and `UART` versions respectively. No need to change the `AFBR_S50_USB` definition as I said above...

mfreed7, 2024-06-03:
> Hi, I just added the missing binary to the [RA4M2 v1.5.6](https://github.com/Broadcom/AFBR-S50-API/releases/download/v1.5.6/AFBR-S50-RA4M2-v1.5.6.zip) archive. It now contains two version of the ExplorerApp, one for USB and one for UART. Please let me know if this fixes your issue.

Awesome! Thank you so much, both for doing that and also pinging me. I'll give it a try.

mfreed7, 2024-06-10:
Thanks again - the new binary seems to work great via UART. I appreciate the help.

closed 2024-06-10 by mfreed7

krishniot, 2025-05-02:
hey @c-berger @mfreed7  ive uploaded the .uart file to the board using rfp-cli 
AFBR.S50.ExplorerApp.v1.5.6_RA4M2_UART.hex
Renesas Flash Programmer CLI V1.12
Module Version: V3.19.00.000
Load: "/home/jchaitanya/Desktop/AFBR.S50.ExplorerApp.v1.5.6_RA4M2_UART.hex" (Size=507323, CRC=D67CB6B5)

Connecting the tool (COM port)
Tool: /dev/ttyACM0
Interface: 2 wire UART

Connecting the target device
Speed: 115,200 bps
Connected to R7FA4M2AD3CFL

Erasing the target device
  [Code Flash 1]       00000000 - 0007FFFF
  [Data Flash 1]       08000000 - 08001FFF                                      
Writing data to the target device                                               
  [Code Flash 1]       00000000 - 0002BF7F
Verifying data on the target device                                             
  [Code Flash 1]       00000000 - 0002BF7F
Writing data to the target device                                               
  [Config Area 1]      0100A100 - 0100A13F
  [Config Area 1]      0100A200 - 0100A2CF                                      
Verifying data on the target device                                             
  [Config Area 1]      0100A100 - 0100A13F
  [Config Area 1]      0100A200 - 0100A2CF                                      
                                                                                

Disconnecting the tool

this was the output afterwards ive connected the jumper to the can vext 
ive tried using example pyscript but still im getting this error
    raise Exception("No data was read from the RX line.")
Exception: No data was read from the RX line.

can you help me out