Broadcom/AFBR-S50-API #30: Incorrect work of excel for Crosstalk Calibration with Explorer

shpegun60, 2024-06-11:
The rows "dS(ext)_corr" and  "dC(ext)_corr" don't have a formula at all, which means I can't create a correct output file

[image]

goetzfrv, 2024-06-12:
Hi shpegun60, sorry for the confusion in the Excel. Actually, most people directly use the API (C-code) version of the xtalk calibration as it does these calculations automatically and you just need to copy paste the xtalk table from your terminal.

In principle you can just copy&paste the values from dS(ext) and dC(ext) into the ..._corr columns and then work with the values in the blue list. Just in case you see spikes in your total xtalk values like shown figure 25 in the AFBR-S50-XTK application note 
[image]

you may need to do an interpolation. 

Please see function `Interpolate_ActivePixels(argus_hnd_t * hnd)` in https://github.com/Broadcom/AFBR-S50-API/blob/main/AFBR-S50/Xtalk-CLI/argus_xtalk_cal_cli.c to see how to do the interpolation (actually averaging).

Best Regards!

goetzfrv, 2024-07-04:
Hello Spehgun60,
please find the corrected Excel under https://github.com/Broadcom/AFBR-S50-API/releases/download/v1.5.6/AFBR-S50-XTK.zip.
Please consider that you might need to manually correct the values in case you need to interpolate because of peaks in the xtalk chart (see my last post). I will close this comment as this will resolve your issues.

closed 2024-07-04 by goetzfrv

shpegun60, 2024-07-04:
Hello [goetzfrv](https://github.com/goetzfrv) 
Thanks for the fixed bug, sorry for not replying, solved the problem with the help of the console program as you recommended to me, thank you very much again!!