Broadcom/AFBR-S50-API #31: Explorer app reference manual is incorrect about the bit width of frame state

mfreed7, 2024-06-16:
The docs for measurement data output (in all modes) say that the "Measurement Frame State Flags" are a HEX16 taking 2 bytes. But in practice (see code links below) that's a 4-byte wide field.

Docs:
https://broadcom.github.io/AFBR-S50-API/explorer_app_cmd_details.html#:~:text=Measurement%20Data%20Set%20(1D%20%2B%203D)%20%2D%20Debug

Code:
https://github.com/Broadcom/AFBR-S50-API/blob/main/Sources/ExplorerApp/api/explorer_api_data.c#L79

The reference for argus_state_t does show 32 bits, but it'd be nice if the tables in the section above reflected that:
https://broadcom.github.io/AFBR-S50-API/group__argus__dca.html#gada1413643a68f3cd36f6c0a1d0b61e91

mfreed7, 2024-06-16:
Fixed in https://github.com/Broadcom/AFBR-S50-API/pull/32

c-berger, 2024-06-18:
Thanks for catching that!

closed 2024-06-18 by c-berger