Broadcom/AFBR-S50-API #34: API Explorer App Reference Manual correction - Measurement Frame State Flags size

essej, 2024-06-30:
In the API App Reference for the Explorer app, in the measurement data commands section that starts here:
   https://broadcom.github.io/AFBR-S50-API/explorer_app_cmd_details.html#cmd_data_full_dbg

The Measurement Frame State Flags field of the message are notated as being HEX16 (2 bytes), but as of the current version of the Explorer API software, that field is actually HEX32 (4 bytes). 

The source code defining that is here:
https://github.com/Broadcom/AFBR-S50-API/blob/6fbd0e1053f602ffc5f8822c6a5a62f54b41b9d0/Sources/ExplorerApp/api/explorer_api_data.c#L79

Just thought it would save some others a bit of time and confusion to have the correct information in the API reference docs! I'm not sure where else to report this discrepancy, apologies if it is the wrong place.

c-berger, 2024-07-01:
Hi there,
thanks for those hints and the detailed explanation.
Error 1 was actually already fixed by #32, just the HTML was not yet updated. Sorry for that inconvenience.
I will fix Error 2 and 3 and let you know once its done.

c-berger, 2024-07-01:
I have updated and fixed the documentation regarding your suggestions.
https://broadcom.github.io/AFBR-S50-API/explorer_app_cmd_details.html#explorer_app_cmd_data
In addition to the issues you mentions, also the ordering of data was wrong. The reference pixels values immediately follow the measuring pixels.

closed 2024-07-01 by c-berger

essej, 2024-07-01:
That's great, thanks!