Broadcom/AFBR-S50-API #37: [build error]unable to build

ksumit12, 2024-07-18:
i am trying to follow the porting mcu guide to implement the API for Nucleo STM32-g474RE board, I have tried it multiple times to follow the porting guide and to build the example_app given in the guide but still getting the same issue. porting guide was actually quite informative but lacks few topic like when you copy \Sources\ExampleApp\main.c into your project's App folder first error you get is Board/Board.h not found and porting guide doesn't mention anything about that similar to this a lot of file were not mentioned in the guide but had issues while compiling for now I copied everything from the sources/platform into my platform ofcourse it won't work since its meant for stm32fxx  this is the error I am getting for now.
` 14:48:01 **** Incremental Build of configuration Debug for project Tof_sensor_example ****
make -j16 all 
arm-none-eabi-gcc "../Platform/driver/flash.c" -mcpu=cortex-m4 -std=gnu11 -g3 -DDEBUG -DUSE_HAL_DRIVER -DSTM32G474xx -c -I../Core/Inc -I../Drivers/STM32G4xx_HAL_Driver/Inc -I../Drivers/STM32G4xx_HAL_Driver/Inc/Legacy -I../Drivers/CMSIS/Device/ST/STM32G4xx/Include -I../Drivers/CMSIS/Include -I"E:/code/Tof_sensor_example/AFBR-S50/Include" -I"E:/code/Tof_sensor_example/AFBR-S50/Test" -I"E:/code/Tof_sensor_example/Platform" -O0 -ffunction-sections -fdata-sections -Wall -fstack-usage -fcyclomatic-complexity -MMD -MP -MF"Platform/driver/flash.d" -MT"Platform/driver/flash.o" --specs=nano.specs -mfpu=fpv4-sp-d16 -mfloat-abi=hard -mthumb -o "Platform/driver/flash.o"
arm-none-eabi-gcc "../Platform/driver/irq.c" -mcpu=cortex-m4 -std=gnu11 -g3 -DDEBUG -DUSE_HAL_DRIVER -DSTM32G474xx -c -I../Core/Inc -I../Drivers/STM32G4xx_HAL_Driver/Inc -I../Drivers/STM32G4xx_HAL_Driver/Inc/Legacy -I../Drivers/CMSIS/Device/ST/STM32G4xx/Include -I../Drivers/CMSIS/Include -I"E:/code/Tof_sensor_example/AFBR-S50/Include" -I"E:/code/Tof_sensor_example/AFBR-S50/Test" -I"E:/code/Tof_sensor_example/Platform" -O0 -ffunction-sections -fdata-sections -Wall -fstack-usage -fcyclomatic-complexity -MMD -MP -MF"Platform/driver/irq.d" -MT"Platform/driver/irq.o" --specs=nano.specs -mfpu=fpv4-sp-d16 -mfloat-abi=hard -mthumb -o "Platform/driver/irq.o"
arm-none-eabi-gcc "../Platform/driver/nvm.c" -mcpu=cortex-m4 -std=gnu11 -g3 -DDEBUG -DUSE_HAL_DRIVER -DSTM32G474xx -c -I../Core/Inc -I../Drivers/STM32G4xx_HAL_Driver/Inc -I../Drivers/STM32G4xx_HAL_Driver/Inc/Legacy -I../Drivers/CMSIS/Device/ST/STM32G4xx/Include -I../Drivers/CMSIS/Include -I"E:/code/Tof_sensor_example/AFBR-S50/Include" -I"E:/code/Tof_sensor_example/AFBR-S50/Test" -I"E:/code/Tof_sensor_example/Platform" -O0 -ffunction-sections -fdata-sections -Wall -fstack-usage -fcyclomatic-complexity -MMD -MP -MF"Platform/driver/nvm.d" -MT"Platform/driver/nvm.o" --specs=nano.specs -mfpu=fpv4-sp-d16 -mfloat-abi=hard -mthumb -o "Platform/driver/nvm.o"
arm-none-eabi-gcc "../Platform/driver/s2pi.c" -mcpu=cortex-m4 -std=gnu11 -g3 -DDEBUG -DUSE_HAL_DRIVER -DSTM32G474xx -c -I../Core/Inc -I../Drivers/STM32G4xx_HAL_Driver/Inc -I../Drivers/STM32G4xx_HAL_Driver/Inc/Legacy -I../Drivers/CMSIS/Device/ST/STM32G4xx/Include -I../Drivers/CMSIS/Include -I"E:/code/Tof_sensor_example/AFBR-S50/Include" -I"E:/code/Tof_sensor_example/AFBR-S50/Test" -I"E:/code/Tof_sensor_example/Platform" -O0 -ffunction-sections -fdata-sections -Wall -fstack-usage -fcyclomatic-complexity -MMD -MP -MF"Platform/driver/s2pi.d" -MT"Platform/driver/s2pi.o" --specs=nano.specs -mfpu=fpv4-sp-d16 -mfloat-abi=hard -mthumb -o "Platform/driver/s2pi.o"
arm-none-eabi-gcc "../Platform/driver/timer.c" -mcpu=cortex-m4 -std=gnu11 -g3 -DDEBUG -DUSE_HAL_DRIVER -DSTM32G474xx -c -I../Core/Inc -I../Drivers/STM32G4xx_HAL_Driver/Inc -I../Drivers/STM32G4xx_HAL_Driver/Inc/Legacy -I../Drivers/CMSIS/Device/ST/STM32G4xx/Include -I../Drivers/CMSIS/Include -I"E:/code/Tof_sensor_example/AFBR-S50/Include" -I"E:/code/Tof_sensor_example/AFBR-S50/Test" -I"E:/code/Tof_sensor_example/Platform" -O0 -ffunction-sections -fdata-sections -Wall -fstack-usage -fcyclomatic-complexity -MMD -MP -MF"Platform/driver/timer.d" -MT"Platform/driver/timer.o" --specs=nano.specs -mfpu=fpv4-sp-d16 -mfloat-abi=hard -mthumb -o "Platform/driver/timer.o"
arm-none-eabi-gcc "../Platform/driver/uart.c" -mcpu=cortex-m4 -std=gnu11 -g3 -DDEBUG -DUSE_HAL_DRIVER -DSTM32G474xx -c -I../Core/Inc -I../Drivers/STM32G4xx_HAL_Driver/Inc -I../Drivers/STM32G4xx_HAL_Driver/Inc/Legacy -I../Drivers/CMSIS/Device/ST/STM32G4xx/Include -I../Drivers/CMSIS/Include -I"E:/code/Tof_sensor_example/AFBR-S50/Include" -I"E:/code/Tof_sensor_example/AFBR-S50/Test" -I"E:/code/Tof_sensor_example/Platform" -O0 -ffunction-sections -fdata-sections -Wall -fstack-usage -fcyclomatic-complexity -MMD -MP -MF"Platform/driver/uart.d" -MT"Platform/driver/uart.o" --specs=nano.specs -mfpu=fpv4-sp-d16 -mfloat-abi=hard -mthumb -o "Platform/driver/uart.o"
../Platform/driver/flash.c: In function 'Flash_Write':
../Platform/driver/flash.c:98:60: error: 'FLASH_SECTOR_1' undeclared (first use in this function); did you mean 'FLASH_SECTOR_SIZE'?
   98 | #define FLASH_CURRENT_SECTOR() ((userConfig[0] == 0x00U) ? FLASH_SECTOR_1 : FLASH_SECTOR_2)
      |                                                            ^~~~~~~~~~~~~~
../Platform/driver/flash.c:150:32: note: in expansion of macro 'FLASH_CURRENT_SECTOR'
  150 |     uint32_t const rd_sector = FLASH_CURRENT_SECTOR();
      |                                ^~~~~~~~~~~~~~~~~~~~
../Platform/driver/flash.c:98:60: note: each undeclared identifier is reported only once for each function it appears in
   98 | #define FLASH_CURRENT_SECTOR() ((userConfig[0] == 0x00U) ? FLASH_SECTOR_1 : FLASH_SECTOR_2)
      |                                                            ^~~~~~~~~~~~~~
../Platform/driver/flash.c:150:32: note: in expansion of macro 'FLASH_CURRENT_SECTOR'
  150 |     uint32_t const rd_sector = FLASH_CURRENT_SECTOR();
      |                                ^~~~~~~~~~~~~~~~~~~~
../Platform/driver/flash.c:98:77: error: 'FLASH_SECTOR_2' undeclared (first use in this function); did you mean 'FLASH_SECTOR_SIZE'?
   98 | #define FLASH_CURRENT_SECTOR() ((userConfig[0] == 0x00U) ? FLASH_SECTOR_1 : FLASH_SECTOR_2)
      |                                                                             ^~~~~~~~~~~~~~
../Platform/driver/flash.c:150:32: note: in expansion of macro 'FLASH_CURRENT_SECTOR'
  150 |     uint32_t const rd_sector = FLASH_CURRENT_SECTOR();
      |                                ^~~~~~~~~~~~~~~~~~~~
../Platform/driver/flash.c:88:9: warning: implicit declaration of function 'FLASH_Erase_Sector'; did you mean 'FLASH_ERASE_SECTOR'? [-Wimplicit-function-declaration]
   88 |         FLASH_Erase_Sector(sector, VOLTAGE_RANGE_3); \
      |         ^~~~~~~~~~~~~~~~~~
../Platform/driver/flash.c:161:5: note: in expansion of macro 'FLASH_ERASE_SECTOR'
  161 |     FLASH_ERASE_SECTOR(wr_sector);
      |     ^~~~~~~~~~~~~~~~~~
In file included from ../Drivers/STM32G4xx_HAL_Driver/Inc/stm32g4xx_hal_def.h:30,
                 from ../Drivers/STM32G4xx_HAL_Driver/Inc/stm32g4xx_hal_rcc.h:27,
                 from ../Core/Inc/stm32g4xx_hal_conf.h:212,
                 from ../Drivers/STM32G4xx_HAL_Driver/Inc/stm32g4xx_hal.h:29,
                 from ../Core/Inc/main.h:30,
                 from ../Platform/driver/flash.c:37:
../Drivers/STM32G4xx_HAL_Driver/Inc/Legacy/stm32_hal_legacy.h:449:39: error: 'FLASH_VOLTAGE_RANGE_3' undeclared (first use in this function); did you mean 'VOLTAGE_RANGE_3'?
  449 | #define VOLTAGE_RANGE_3               FLASH_VOLTAGE_RANGE_3
      |                                       ^~~~~~~~~~~~~~~~~~~~~
../Platform/driver/flash.c:88:36: note: in expansion of macro 'VOLTAGE_RANGE_3'
   88 |         FLASH_Erase_Sector(sector, VOLTAGE_RANGE_3); \
      |                                    ^~~~~~~~~~~~~~~
../Platform/driver/flash.c:161:5: note: in expansion of macro 'FLASH_ERASE_SECTOR'
  161 |     FLASH_ERASE_SECTOR(wr_sector);
      |     ^~~~~~~~~~~~~~~~~~
../Drivers/STM32G4xx_HAL_Driver/Inc/Legacy/stm32_hal_legacy.h:421:39: error: 'FLASH_TYPEPROGRAM_WORD' undeclared (first use in this function); did you mean 'FLASH_TYPEPROGRAM_FAST'?
  421 | #define TYPEPROGRAM_WORD              FLASH_TYPEPROGRAM_WORD
      |                                       ^~~~~~~~~~~~~~~~~~~~~~
../Platform/driver/flash.c:165:27: note: in expansion of macro 'TYPEPROGRAM_WORD'
  165 |     if (HAL_FLASH_Program(TYPEPROGRAM_WORD, wr_sector_addr, 0) != HAL_OK)
      |                           ^~~~~~~~~~~~~~~~
[... cut ...]
another thing I want to ask is I somehow made my own PCB similar to the BDC-AFBR-S50 TOF SENSOR BOARD although I haven't tested that pcb yet but I plan to use CAN_App directly on that to get the readings in the CAN . i will be used it with ardupilot and it that for most peripheral it uses droneCAN which is like a wrapper for CAN I think but there is a firmware for that which does that for you I was thinking can I combine that with CAN_App so that I can directly get DroneCAN output? please suggest a way for that if it's possible. [https://github.com/ArduPilot/ardupilot/tree/master/Tools/AP_Periph](url) (link of the firmware that I was thinking to combine)

goetzfrv, 2024-07-18:
Hello ksumit12,
sorry that the porting troubles you so much. If you don't have so much experience with working with st MCUs and the STMcube IDE but you are actually using a ST MCU it is most convenient to just use the preset ST example from our repository and just import it to your IDE. This guarantees that you already have the correct file structure set up. Just download or clone the repo and choose the STM example project.
[image]

ksumit12, 2024-07-18:
hey, thanks for quick reply for my application I actually made my own PCB similar to BDC-AFBR-S50 TOF SENSOR BOARD since I need output in CAN bus that can be directly used in ardupilot so eventually I will be using RA4M2 board till then I wanted to test the module with nucleo-stm32 g474re. just to try I builded the AFBR_S50_Example_F401RE and try to upload this on my board it got flashed but didn't worked obviously , do you have any solutions for that? i don't have f401re board with me so I can't directly try the provided code.

goetzfrv, 2024-07-18:
Well, assuming you have successfully migrated the MCU from the R401RE example to your MCU and changed the pin & interface definitions and timers (if necessary) accordingly, you may need to specify 'did not work'. Do you get any output on the terminal? If no check your UART implementation/porting. If yes and you failed the HAL test you may have issues with e.g. the SPI interface. For that and other issues see - https://broadcom.github.io/AFBR-S50-API/faq.html

c-berger, 2024-07-18:
The issue is about the "flash" module. It is actually an optional module that is used to persistently store values on the MCUs flash. I a first attempt, I'd recommend to disable this (i.e. just don't compile the flash and nvm files in the platform layer) to see if it is compiling correctly. In a second step, if the usage of persistent calibration data is required, you can enable it again and define the required flash sections that are used by the module in the linker script.

These are the files to exclude from the build:
[image]

This is the line in question in the flash.c file. Note the `userConfig` should get mapped into the `user_data` section in flash which needs to be defined:
[image]

And here is how the `user_data` it is currently defined in the linker script (STM32F401RETX_FLASH.ld)
[image]

closed 2024-09-10 by c-berger