Broadcom/AFBR-S50-API #40: Unclear what Signal Quality of Measurement Result is Based on

dmammolo, 2025-01-07:
Dear Broadcom,

I am wondering what the `SignalQuality` entry in `argus_results_bin_t` is based on as I could not find any description to that in the API or any document provided by Broadcom. For our usecase it would be important to know how confident we can be about the correctness of the result, thus I have following questions/observation that would be nice to have answered:
- We observed that the `SignalQuality` is either 1 or 100 and never something in between although it's described as percentage, how is that?
- How is the quality computed? Is it based on the setup of the Pixel Binning Algorithm? If yes what leads to 1 or 100?
- Is it possible to get a completely wrong range estimate although the SignalQuality is at 100? E.g. due to using a cover glass and not calibrating it?

goetzfrv, 2025-01-08:
Dear dmammolo,
thanks for your comment and request. Indeed until and including release v1.5.6 the signal quality indicator is just implemented as coarse figure. When being "100" the frame status is actually "0" (i.e. ok). A SQI  of "1" simply indicates that there is a warning which is a status > 0 and SQI="0" means that there is an error by means of the status < 0. In the next release 1.6.5 which is currently on release stage we have refined the indicator considering a signal-to-noise (SNR) as baseline. You can expect the release by end of this month.

dmammolo, 2025-01-08:
Thank you for the quick answer. In that case I am looking forward to the new release!

closed 2025-01-08 by dmammolo