Broadcom/AFBR-S50-API #41: Low distance measurement

FloVesuv, 2025-02-03:
Hello,  

I am using the AFBR-S50MV85I sensor for measurements within a range of 50 to 150mm (black surface to be measured).  

I get good results around calibration, however, when I move away from it, I can have significant offsets (for example, with calibration at 100mm and a measurement at 60mm, I measure 70mm) with the measurement provided by the pixel binning algorithm.  

Is this due to the sensor's limitations at short distances?  

I am trying to work with the raw values of each pixel independently to weight them; some are quite close to the result (60mm), while others are completely out of tolerance.  
My goal is not to redo the pixel binning algorithm but to develop my own in order to get more relevant results with the actual measurement.

What data could be useful for performing this weighting?  
Can the amplitude res->Pixels[i].Amplitude be useful to me?  
I saw that it is used in the PBA, but some information is missing, such as the "AbsoluteMinimumDistanceThreshold".

Regards

goetzfrv, 2025-02-04:
Hello FloVesuv,
thanks for your information. It is true that the sensor's absolute distance shows some offsets on the near ranges below 150mm. This is due to the parallax effect and the associated transition to other illuminated pixels. Even though there is a default offset table for the single pixel offsets there are some deviations which need to be compensated for highest accuracy needs. 

For pixel-to-pixel offset calibration it is advisable to perform the see https://broadcom.github.io/AFBR-S50-API/group__argus__meas.html#gabfbc2d139e090be5e06930814c4bf48e (Argus_ExecuteRelativeRangeOffsetCalibrationSequence())

For highest accuracy needs you may need to create a look-up table and collect all offsets of all ranges for each device under test. 

For near range measurements we usually suggest the 85G variant which has a decent trade-off between limited number of illuminated pixels and enough pixels to compensate pixel-to-pixel offsets.

Best regards!

goetzfrv, 2025-02-13:
Yes, for highest absolute accuracy for these close ranges you may need to have a look-up table.
Even though the 85G is the best option, you still may need a look-up table for these ranges.

Regards,
ToF support

goetzfrv, 2025-03-28:
Hello Flo,
if not done yet please also try your near range tests with the latest release v1.6.5 as the offsets should have been further improved. I will close this topic by now but feel free to re-open if you still have issues.

closed 2025-03-28 by goetzfrv