Broadcom/AFBR-S50-API #44: AFBR S50 Explorer v1.6.5 on Windows does not want to connect to COM Port

emelin2004, 2025-03-21:
Hi everyone, as I have already mention in a title, the AFBR Explorer on Windows does not want to communicate over COM port.

My Setup: 

- Windows 11 Enterprise 23H2
- Pi Pico running Latest Picoprobe Firmware
- AFBR S50 Refernce Board based on NXP MKL17Z256

My Problem: 
- Using your provided Python SCI Example the Python app is able to Communicate with the Sensor board over the Serial Port the Pico creates and it returns and Prints proper data
- But When using AFBR Explorer the GUI for some reason can not communicate with the Serial Port the Pico creates.

Connecting Sensor board over separate FTDI Cable does work and the Explorer is able to communicate with the App.

What have I tried for now?

- I analysed the UART connection between the Board and the Pico when running [SCI Python example](https://broadcom.github.io/AFBR-S50-API/sci_python_example_8py-example.html) on my windows pc and the data is flowing as it should as described [here ](https://broadcom.github.io/AFBR-S50-API/explorer_app_cmd_overview.html). 
- When trying to connect to this COM Port using the Explorer the app yields an error. When looking at the UART Bus on my Oscilloscope there is no data flowing in any direction when the GUI tries to iniate the connection. When running the Python script the data is flowing in both directions.

This is a Software Bug in the AFBR Explorer GUI. Connecting to the COM Port using Putty or any other Terminal returns no error and data can be sent to board and I did confirm that by analysing the UART bus.

I have even tried other Windows devices and I get the same result.

Here is the Log of what happens when trying to connect: using GUI

`2025-03-21 15:37:31.586 # 1   Debug: USB: searching available devices w/ GUID: "d1a1f25e-581d-438b-857f-16fd247554ec"
2025-03-21 15:37:31.586 # 1   Debug: USB: found 0 devices:
                                      -
2025-03-21 15:37:31.586 # 1   Debug: DataLink: Searching for COM Ports
2025-03-21 15:37:31.837 # 1   Debug: DataLink: found 3 devices:
                                      - COM5
                                      - COM4
                                      - COM19
2025-03-21 15:37:33.225 # 1   Debug: DataLink: Searching for COM Ports
2025-03-21 15:37:33.467 # 1   Debug: DataLink: found 1 devices (path: "\\?\serial#COM19#115200"):
                                      - COM19
2025-03-21 15:37:33.467 # 1   Debug: SCI: Create a new Serial Data Link connection (device path: "\\?\serial#COM19#115200").
2025-03-21 15:37:33.467 # 1    Info: Data Link (Serial) #1: Opening connection to \\?\serial#COM19#115200.
2025-03-21 15:37:33.467 # 1    Info: Data Link (Serial) #1: Opening serial connection to COM19 @ 115200 bps.
2025-03-21 15:37:33.467 # 1    Info: Data Link (Serial) #1: Opened port COM19 @ 115200 bps
2025-03-21 15:37:33.712 # 1   Debug: SCI: Ping Failed on Data Link (Serial) #1
                                     Sending frame yielded timeout (CMD: Ping (0x01)).
                                        bei AFBR.S50.SysCom.SCI.SendFrame(DataLink dataLink, TxFrame frame)
                                        bei AFBR.S50.SysCom.SCIConnectionManager.TestConnection(DataLink dataLink)
2025-03-21 15:37:33.712 # 1    Info: Data Link (Serial) #1: Connection is not OK on COM19 @ 115200 bps
2025-03-21 15:37:33.712 # 1    Info: Data Link (Serial) #1: Closing connection "".
2025-03-21 15:37:33.712 # 1    Info: Data Link (Serial) #1: Opening serial connection to COM19 @ 1000000 bps.
2025-03-21 15:37:33.712 # 1    Info: Data Link (Serial) #1: Opened port COM19 @ 1000000 bps
2025-03-21 15:37:33.759 # 1   Debug: SCI: Ping Failed on Data Link (Serial) #1
                                     Sending frame yielded timeout (CMD: Ping (0x01)).
                                        bei AFBR.S50.SysCom.SCI.SendFrame(DataLink dataLink, TxFrame frame)
                                        bei AFBR.S50.SysCom.SCIConnectionManager.TestConnection(DataLink dataLink)
2025-03-21 15:37:33.759 # 1    Info: Data Link (Serial) #1: Connection is not OK on COM19 @ 1000000 bps
2025-03-21 15:37:33.759 # 1    Info: Data Link (Serial) #1: Closing connection "".
2025-03-21 15:37:33.759 # 1    Info: Data Link (Serial) #1: Opening serial connection to COM19 @ 500000 bps.
2025-03-21 15:37:33.759 # 1    Info: Data Link (Serial) #1: Opened port COM19 @ 500000 bps
2025-03-21 15:37:33.838 # 1   Debug: SCI: Ping Failed on Data Link (Serial) #1
                                     Sending frame yielded timeout (CMD: Ping (0x01)).
                                        bei AFBR.S50.SysCom.SCI.SendFrame(DataLink dataLink, TxFrame frame)
                                        bei AFBR.S50.SysCom.SCIConnectionManager.TestConnection(DataLink dataLink)
2025-03-21 15:37:33.838 # 1    Info: Data Link (Serial) #1: Connection is not OK on COM19 @ 500000 bps
2025-03-21 15:37:33.838 # 1    Info: Data Link (Serial) #1: Closing connection "".
2025-03-21 15:37:33.838 # 1    Info: Data Link (Serial) #1: Opening serial connection to COM19 @ 2000000 bps.
2025-03-21 15:37:33.838 # 1    Info: Data Link (Serial) #1: Opened port COM19 @ 2000000 bps
2025-03-21 15:37:33.886 # 1   Debug: SCI: Ping Failed on Data Link (Serial) #1
                                     Sending frame yielded timeout (CMD: Ping (0x01)).
                                        bei AFBR.S50.SysCom.SCI.SendFrame(DataLink dataLink, TxFrame frame)
                                        bei AFBR.S50.SysCom.SCIConnectionManager.TestConnection(DataLink dataLink)
2025-03-21 15:37:33.886 # 1    Info: Data Link (Serial) #1: Connection is not OK on COM19 @ 2000000 bps
2025-03-21 15:37:33.886 # 1    Info: Data Link (Serial) #1: Closing connection "".
2025-03-21 15:37:33.886 # 1    Info: Data Link (Serial) #1: Failed to open a serial connection on COM19!
                                     The following baud rates were attempted: 115200 bps, 1000000 bps, 500000 bps, 2000000 bps
2025-03-21 15:37:33.886 # 1    Info: Data Link (Serial) #1: Open connection failed!
                                     Data Link (Serial) #1: Failed to open a serial connection on COM19!
                                     The following baud rates were attempted: 115200 bps, 1000000 bps, 500000 bps, 2000000 bps
                                        bei AFBR.S50.SysCom.SerialDataLink.OnOpen(String portName, Int32 baudRate, Predicate`1 testConnection)
                                        bei AFBR.S50.SysCom.SerialDataLink.OnOpen(DataLinkConnectionInfo connectionInfo, Predicate`1 testConnection)
                                        bei AFBR.S50.SysCom.DataLink.Open(DataLinkConnectionInfo connectionInfo, Predicate`1 testConnection)
                                        bei AFBR.S50.SysCom.SCIConnectionManager.GetDataLink(DataLinkConnectionInfo connection, Predicate`1 testConnection)
2025-03-21 15:37:33.886 # 1    Info: Data Link (Serial) #1: Closing connection "".
2025-03-21 15:37:33.886 # 1    Info: Data Link (Serial) #1: Closing connection "".
2025-03-21 15:37:33.886 # 1   Error: Device ID 1: Open auto-connection failed.
                                     Failed to open connection to "\\?\serial#COM19#115200", address 0
                                        bei AFBR.S50.SysCom.SCIConnectionManager.OpenDataLink(String devicePath, Byte deviceAddress)
                                        bei AFBR.S50.SysCom.SCI.OpenConnection(String devicePath, Byte deviceAddress)
                                        bei AFBR.S50.Controller.ArgusController.OpenConnectionToPath(String devicePath, Byte deviceAddress)
                                        bei AFBR.S50.GUI.Controller.ArgusControllerViewModel.Connect(String devicePath, Byte deviceAddress)
                                     --------------------`

c-berger, 2025-03-21:
Hi, thanks for the detailed explanation. Looks like you have already done a good job and I am sorry for the inconvenience.
The auto-connection mechanism was designed with only our boards in mind. So it probably simply doesn't recognize your individual Pico board. 
From the log, one can see that there are 3 available COM ports (4, 5 and 19) while the GUI only attempts to connect to COM19.
You can manually select a different COM port and skip the auto-connection mechanism via Menu > Device > Connect and then select the COM port and baud rate manually.

emelin2004, 2025-03-21:
> Hi, thanks for the detailed explanation. Looks like you have already done a good job and I am sorry for the inconvenience.
> The auto-connection mechanism was designed with only our boards in mind. So [...]

Hi, firstly thanks for your fast response!

I forgot to note that I have already tried that and that yields the same result.

COM19 is in my case the right port, again, confirmed with your Python SCI script.

If I can somehow provide you with more info then please tell me how. 

Thanks.

emelin2004, 2025-03-21:
I think i have found out what the problem is! I tried to create a simple .NET app to send few bytes over this UART and I noticed data was not getting sent. 

The fix was to set DtrEnable on SerialPort to true!

After modifying the GUI APP and doing the same I was able to make the GUI App communicate with the Sensor over Picoprobe COM Port!

Like this:
`
this.SerialPort.Open();
this.SerialPort.DtrEnable = true;`

I think this could be implemented into GUI by default to fix incompatibility with some Serial Ports.

c-berger, 2025-03-24:
Thanks for your solution. That's a quite interesting finding!

The [DtrEnable](https://learn.microsoft.com/en-us/dotnet/api/system.io.ports.serialport.dtrenable?view=net-9.0-pp) is used to enable hardware or software handshake which we don't use on our boards. It probably will break the compatibility with our boards if we enable it permanently. 
Thus, it might be worth it to disable handshaking on the Pico side instead. I'd be a little concerned about the implications as we have never tested with handshaking. 
Btw., handshaking is implemented on a higher, hardware independent layer, in our SCI protocol already...

Anyway, I agree that we probably could add an options dialog to tweak the serial configuration. Meanwhile, I'd recommend sticking to the current configuration.

emelin2004, 2025-03-24:
> Thanks for your solution. That's a quite interesting finding!
>
> The [DtrEnable](https://learn.microsoft.com/en-us/dotnet/api/system.io.ports.serialport.dtrenable?view=net-9.0-pp) is used to enable [...]

Thanks for your reply! I am now going to mark this issue as closed.

closed 2025-03-24 by emelin2004