Broadcom/AFBR-S50-API #46: UART Interface Test Failed: "No data was read from the RX line."

WilliamChung8910, 2025-05-13:
Description:

I'm using the AFBR-S50 Reference Board by MikroElektronika.

Firmware: AFBR.S50.ExplorerApp.v1.6.5_RA4M2_UART.hex

Connection: Connected to PC via USB using FT232RL USB to UART module

Environment: Running the Python UART interface example provided

When I run the Python example script (sci_python_example_raw.py), I get the following output and error:
```
AFBR-S50: Open Serial Port /dev/ttyUSB0
AFBR-S50: Serial Port is open /dev/ttyUSB0: True
setting data output mode to 1d data only
Sending: 024107f503
Traceback (most recent call last):
  File "sci_python_example_raw.py", line 247, in <module>
    s50.write(bytes.fromhex("02 41 07 F5 03"))
  File "sci_python_example_raw.py", line 97, in write
    return self.__wait_for_ack(tx[1])  # read acknowledge
  File "sci_python_example_raw.py", line 115, in __wait_for_ack
    raise Exception("No data was read from the RX line.")
Exception: No data was read from the RX line.
```

Confirmed that the correct serial port (/dev/ttyUSB0) is selected. Baudrate 115200
FT232 using 5V
Tried v1.4.4 v1.5.6 v1.6.5. problem is same 

checked #29 but still cannot receive data
Any help or insights would be greatly appreciated. Thank you!

AminAliz, 2025-06-01:
Hi William.
Did you design the FT232 circuit yourself or did you use a module? If you designed it yourself, please send the schematic file. If you used a module, please send the module specifications.

goetzfrv, 2025-06-16:
Hello William,
if the issue still persists please post the code you are using. Just end of last year we updated the script because we saw an issue with the  default baud rate and thus fixed it to be 1Mbps. Please also make sure you have correctly swapped Tx<->Rx communicating at the UART inferfaces. Please also do let us know if it is working because then we'll close this topic.
Thanks

goetzfrv, 2025-07-24:
Issue closed because of no response.

closed 2025-07-24 by goetzfrv